evilsocket/pwnagotchi #388: [BUG] ImportError: cannot import name 'monitoring' from 'Tensorflow.python.eager'

mlinton, 2019-10-24:
Attempting to use the v1.0.1 code on a generic debian platform:

After downloading and attempting to install with "sudo pip3 install ." I get an error regarding the tensorflow python libraries after it states that it was successfully build pwnagotchi:

"tensorflow 1.13.1 has requirement tensorflow-estimator<1.14.0rc0,>=1.13.0, but you'll havee tensorflow-estimator 1.14.0 which is incompatible.

When pwnagotchi is then started it throws an error on startup:

"Unhandled exception in thread started by <bound metehod AsyncTrainer._ai_worker of <pwnagotchi.agent.Agent object at 0x7f226628bda0>"

And an ImportError: cannot import name 'monitoring' from 'tensorflow.python.eager'

I thought it was maybe a conflict with other software so I attempted to remove and re-install the python libraries with pip unsuccessfully.

dadav, 2019-10-25:
Whats your `pip3 freeze`?

mlinton, 2019-10-25:
absl-py==0.8.1
asn1crypto==0.24.0
astor==0.8.0
Brlapi==0.6.7
certifi==2018.8.24
chardet==3.0.4
chrome-gnome-shell==0.0.0
Click==7.0
cloudpickle==1.2.2
crypto==1.4.1
cryptography==2.6.1
cupshelpers==1.0
cycler==0.10.0
distro==1.3.0
distro-info==0.21
dnspython==1.16.0
entrypoints==0.3
file-read-backwards==2.0.0
Flask==1.1.1
future==0.17.1
gast==0.2.2
grpcio==1.24.3
gym==0.14.0
h5py==2.10.0
httplib2==0.11.3
idna==2.6
impacket==0.9.20.dev0
inky==0.0.5
iotop==0.6
itsdangerous==1.1.0
Jinja2==2.10.1
joblib==0.14.0
Keras-Applications==1.0.8
Keras-Preprocessing==1.1.0
keyring==17.1.1
keyrings.alt==3.1.1
kiwisolver==1.1.0
ldap3==2.5.1
ldapdomaindump==0.9.1
louis==3.8.0
Mako==1.0.7
Markdown==3.1.1
MarkupSafe==1.1.0
matplotlib==3.1.1
mock==3.0.5
mpi4py==3.0.2
Naked==0.1.31
numpy==1.17.2
oauthlib==3.1.0
olefile==0.46
opencv-python==4.1.1.26
pandas==0.25.2
Pillow==5.4.1
protobuf==3.10.0
pwnagotchi==1.0.1
pyasn1==0.4.7
pycairo==1.16.2
pycrypto==2.6.1
pycryptodomex==3.9.0
pycups==1.9.73
pycurl==7.43.0.2
pyglet==1.3.2
PyGObject==3.30.4
pyOpenSSL==19.0.0
pyparsing==2.4.2
PySimpleSOAP==1.16.2
pysmbc==1.0.15.6
PySocks==1.7.1
python-apt==1.8.4
python-dateutil==2.8.0
python-debian==0.1.35
python-debianbts==2.8.2
pytz==2019.3
pyxdg==0.25
PyYAML==5.1
reportbug===7.5.3-deb10u1
requests==2.21.0
requests-oauthlib==1.2.0
RPi.GPIO==0.7.0
scapy==2.4.3
scipy==1.3.1
SecretStorage==2.3.1
shellescape==3.4.1
six==1.12.0
smbus2==0.3.0
spidev==3.4
stable-baselines==2.7.0
tensorboard==1.13.1
tensorflow==1.13.1
tensorflow-estimator==1.14.0
termcolor==1.1.0
tweepy==3.6.0
unattended-upgrades==0.1
urllib3==1.24.1
Werkzeug==0.15.6

mlinton, 2019-10-25:
Should I be using pyenv or something?

mlinton, 2019-10-25:
[image]

dadav, 2019-10-25:
I always recommend to install stuff in virtualenvs, but i don't think this is the problem here.

mlinton, 2019-10-25:
Downgraded the tensorflow_estimator installation to tensorflow-estimator 1.13.0 and now it no longer throws the errors.

mlinton, 2019-10-25:
We should probably leave this open until we can confirm that a fix isn't required.

hectorm, 2019-12-08:
Hi, I confirm that after downgrading to `tensorflow-estimator 1.13.0`, my problem has been fixed.

I have noticed that in PyPI `tensorflow 1.13.1` has as dependency `tensorflow-estimator (<1.14.0rc0, >=1.13.0)` while in PiWheels (the repository that Raspbian uses) has `tensorflow-estimator (<1.15.0rc0, >=1.14.0rc0)`.

It seems that the correct value is the one of PyPi [according to the source code of TensorFlow](https://github.com/tensorflow/tensorflow/blob/v1.13.1/tensorflow/tools/pip_package/setup.py#L60).

hectorm, 2019-12-10:
I have opened an issue regarding this problem in piwheels/packages#66.