evilsocket/pwnagotchi #390: Can't rotate 2.7 waveshare screen image

talonawd91, 2019-10-25:
Trying to use the "rotation" statement within config.yml setting the display to "rotation: 180". This has no effect on the image. 

**To Reproduce**
Change display section in config.yml:
ui:
   fps: 0.0
   display:
      type: 'waveshare27inch'
      color: 'black'
      rotation: 180

**Expected behavior**
screen should rotate 180 from current orientation. 

**Environment (please complete the following information):**
 - Pwnagotchi version:
(v1.1.0b)
 - OS version
10.1
 - Type of hardware
Pi 3b+
 - Any additional hardware used
No

dadav, 2019-10-25:
180 is the default value. Maybe try to set it to something different like 90. Still no rotation?

Frituurpanda, 2019-10-25:
@dadav @talonawd91 almost correct, ```rotation: 0``` will do the trick ;).

dadav, 2019-10-25:
@talonawd91 can you confirm? :)

talonawd91, 2019-10-26:
I just changed it. Works like a charm. Thanks for the guidance.

closed 2019-10-26 by talonawd91