evilsocket/pwnagotchi #391: [BUG] Auto-Backup fails if one of the files is missing

deveth0, 2019-10-25:
**Describe the bug**
I'm not sure, how this situation came up but I don't have a `.api-report.json` file in `/root` and the auto-backup plugin therefor fails.
Probably it should check for each file first and only backup those files, that exist?
Other question: What should the `api-report.json` contain?

**To Reproduce**

1. delete `/root/.api-report.json`
2. activate auto-backup
3. see failure

**Expected behavior**
Only existing files are backuped

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.0.1

**Additional context**
Add any other context about the problem here.

dadav, 2019-10-25:
Well, the plugin just executes the commands its given. We should change the default files to files which are 100% there. Api report is only there if the grid plugin is enabled..

deveth0, 2019-10-25:
thx for the info. I created a PR which adds a simple check, if the file exists. I think this is a clear solution.

closed 2019-10-25 by evilsocket