evilsocket/pwnagotchi #394: [BUG] pip3 install fail on invalid syntax

Xoib, 2019-10-25:
Trying to install on a Pi Model 3B not through the image but manually.

```
sudo pip3 install --no-cache-dir . 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/pwnagotchi-1.0.1
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-nxh8ujrt/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-nxh8ujrt/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-req-build-nxh8ujrt/pip-egg-info
         cwd: /tmp/pip-req-build-nxh8ujrt/
    Complete output (12 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-nxh8ujrt/setup.py", line 4, in <module>
        import pwnagotchi
      File "/tmp/pip-req-build-nxh8ujrt/pwnagotchi/__init__.py", line 5, in <module>
        import pwnagotchi.ui.view as view
      File "/tmp/pip-req-build-nxh8ujrt/pwnagotchi/ui/view.py", line 7, in <module>
        import pwnagotchi.utils as utils
      File "/tmp/pip-req-build-nxh8ujrt/pwnagotchi/utils.py", line 179
        bpf_filter = " or ".join([f"wlan type mgt subtype {subtype}" for subtype in subtypes])
                                                                   ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

dadav, 2019-10-25:
What python version are you using? You need 3.6+

Xoib, 2019-10-25:
Thanks, I missed that. I have Python 3.5.
I am installing Python 3.8. Well currently compiling because it takes time.

```
 wget https://www.python.org/ftp/python/3.8.0/Python-3.8.0.tgz
 tar xf Python-3.8.0.tgz
 cd Python-3.8.0
 ./configure --enable-optimizations
 make -j 4
 sudo make altinstall
```

closed 2019-10-25 by Xoib