evilsocket/pwnagotchi #397: Options in /etc/pwnagotchi/config.yml isn't beeing applied

joseaguardia, 2019-10-25:
Hi.

I'm using latest version of official raspberryPI image running in RPi ZeroW.

I've created the /etc/pwnagotchi/config.yml file with tha below content, However, it seems like that config file isn't been loaded, because I get the handshake of whitelisted SSID (wlan_home2G) and my custom plugin doesn't run (if I overwrite a default plugin, my script works fine)

`main:
    custom_plugins: /opt/plugins
    plugins:
      #conexion bluetooth al movil
      bt-tether:
        enabled: true
        mac: 'A8:DA:75:00:8F:AA' 
        ip: '192.168.44.44' 
        netmask: 24
        interval: 1
        share_internet: true   # this will change the routing and nameserver on your pi

      memtemp: # Display memory usage, cpu load and cpu temperature on screen
        enabled: true
        orientation: horizontal

      # access points to ignore
      whitelist:
          - wlan_home2G
         
      convertir:
        running: true

ui:
  display:
    video:
      address: 0.0.0.0`

Regards.

Thanks

joseaguardia, 2019-10-25:
If I run the command manually with the -C option:

[image]

xenDE, 2019-10-25:
the FAY says, whitelistet SSIDs will not be deauthed.

if a device connects to your ssid and a handshake is collected, it will be saved.

https://pwnagotchi.ai/faq/#how-do-i-whitelist-my-home-network-so-pwnagotchi-stops-pwning-me

dadav, 2019-10-25:
Derek close

closed 2019-10-25 by derek[bot]