evilsocket/pwnagotchi #403: Include plugin data in the backup files

emedvedev, 2019-10-26:
**Is your feature request related to a problem? Please describe.**
Every time I reflash my pwnagotchi and restore it from backup, the plugins such as OHC begin to upload my handshakes (several hundred of them) all over again, because the list of already uploaded handshakes is not backed up.

**Describe the solution you'd like**
I would like to move uploaded lists to `/etc/pwnagotchi` from `/root`, as it will be included in the backup script then.

**Describe alternatives you've considered**
I've also considered specifically including the files in the backup script and `defaults.yml`, but this looks clumsy. I'd rather have a specific directory to store such things, have it backed up, and have people use it when developing new plugins. `/etc/pwnagotchi` is a good fit semantically, imo.

**Additional context**

1. I have a branch for that and can start a PR, but wanted to raise an issue as per guidelines to discuss it first.
2. It would be great to backup quickdic's `/opt/wordlists` as well, or maybe move it to `/etc/pwnagotchi/wordlists`, as it's also something I have to restore every time. Both would make sense, and the wordlists Pwnagotchi can handle are pretty small, so it won't be a significant increase to the backup size.

evilsocket, 2019-10-26:
what prevents you from adding /root/.* to the backup list?

emedvedev, 2019-10-26:
Nothing. :) I can submit a PR adding `/root` if you prefer, but I think that people who do a custom Pwnagotchi installation instead of flashing the .img might have all sorts of unrelated stuff in `/root`. If we move the files to `/etc/pwnagotchi` and back it up, we can be sure that there's only relevant files.

e-minguez, 2019-10-27:
Maybe it would be nice for all plugins to store their config & stuff in a normalized folder (such as /root/pwnagotchi-plugins/whatever or /opt/pwnagotchi-plugins/stuff...) so the backup script can just backup those, no matter what plugins are used and won't backup unnecessary stuff.

emedvedev, 2019-10-27:
@e-minguez #420 proposes to store the persistent configs in `/etc/pwnagotchi`.

evilsocket, 2019-10-27:
this is not a bug and putting status files in /etc/ is against the POSIX standard. They should be left where they are, or in /var/run/

personal preference != bug

closed 2019-10-27 by evilsocket