evilsocket/pwnagotchi #406: On 1.1.0 RC0 fresh install hosts and hostname are both set to "pwnagotchi" no matter what you place as your name in the config.yml

Arttumiro, 2019-10-26:
I dont know if it matters too much since it is simple to change it but still, i didnt try manually rebooting and seeing if that changed things.

Arttumiro, 2019-10-26:
There is the log file
[pwnagotchi.log](https://github.com/evilsocket/pwnagotchi/files/3774967/pwnagotchi.log)

evilsocket, 2019-10-26:
from this log i can see your unit's hostname is `Kattichu` ... what's the issue?

Arttumiro, 2019-10-26:
I changed that manually, i thought that making "name: 'kattichu' " in the config.yml would change it automatically, either way im just being dumb.

closed 2019-10-26 by Arttumiro

evilsocket, 2019-10-26:
i needed the logs before you changed it manually, the very first boot ...

234v244, 2019-12-01:
I have a similar problem, and don't know how to fix it.