evilsocket/pwnagotchi #407: Version 1.0.1RC0 up timer and rpi3 on manual mode after shutdown

Arttumiro, 2019-10-26:
Used sudo shutdown now, unplugged and plugged the rpi3 back in to power, it went into manual and the up time is all bugged out
[image]

Arttumiro, 2019-10-26:
Wait.. i mean version 1.1.0, sorry

Arttumiro, 2019-10-26:
I got it into auto mode by doing the command that appears when you ssh into it, but always on power on its on manual when using a raspberry pi 3 atleast

evilsocket, 2019-10-26:
weird, the uptime is read from  `/proc/uptime`, can you cat it when this happens and paste it here?

Arttumiro, 2019-10-26:
Its still going to manual mode but the uptime bug is not happening anymore

closed 2019-10-26 by evilsocket