evilsocket/pwnagotchi #431: [BUG] Error when running win_connection_share.ps1 script. "An event was unable to invoke any of the subscribers"

w0ntonsoup, 2019-10-27:
**Describe the bug**
When attempting to run the `win_connection_share.ps1 -EnableInternetConnectionSharing` command, I get the following error:

```
C:\win_connection_share.ps1 : An event was unable to invoke any of the subscribers (Exception
from HRESULT: 0x80040201)
At line:1 char:1
+ .\win_connection_share -EnableInternetConnectionSharing
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (:) [win_connection_share.ps1], COMException
    + FullyQualifiedErrorId : System.Runtime.InteropServices.COMException,win_connection_share.ps1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure pwnagotchi is available and alive. SSH into pwnagotchi on `10.0.0.2` to ensure connectivity.
2. Download `win_connection_share.ps1` from `script` folder within repo
3. Run `win_connection_share.ps1 -EnableInternetConnectionSharing`.

Extra step: I've rebooted my computer to see if that would change any outcome.

**Expected behavior**
To enable internet connection sharing from my Windows host to my Pwnagotchi.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version: v1.1.0RC0
 - OS version: Windows 10
 - Raspberry Pi Zero W

w0ntonsoup, 2019-10-28:
I've found that Windows ICS (Internet Connection Sharing) was the culprit here. I restarted the ICS service by following these steps:

1. Click Windows Start > Control Panel.
2. Choose `Administrative Tools`
3. Choose `Services`
4. Scroll down the services window and look for `Internet Connection Sharing`.
5. Right-click on the Internet Connection Sharing item, and then select `Restart`

After these steps, run the script again with the `-EnableInternetConnectionSharing` flag and you should see that ICS is enabled.

closed 2019-10-28 by w0ntonsoup

mrseeker, 2019-11-04:
I tried the issues suggested here, did not work. In the end, I found out that if you go to "network connections", rightclick on the connection (in my case the wireless one), Settings, Sharing, and then select the RNDIS network, then it will work.

cestodd, 2020-11-20:
> I found out that if you go to "network connections", rightclick on the connection (in my case the wireless one), Settings, Sharing, and then select the RNDIS network, then it will work.

> you save my day , thx alot.