evilsocket/pwnagotchi #437: Option for 'fast' or 'traditional' Inky rendering

dipsylala, 2019-10-27:
Some users have expressed a preference for using the old Inky refresh due to artifacts or in one case burn-in.

**Describe the solution you'd like**
A configuration setting to set whether Inky or InkyFast is used. The change is limited to one area:

https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ui/hw/inky.py#L34

```
        from pwnagotchi.ui.hw.libs.inkyphat.inkyphatfast import InkyPHATFast
        self._display = InkyPHATFast(self.config['color'])
        self._display.set_border(InkyPHATFast.BLACK)
```

or:

```
        from inky import InkyPHAT
        self._display = InkyPHAT(self._display_color)
        self._display.set_border(InkyPHAT.BLACK)
```

A different display type would complicate things, so possibly an inky-specific config or specific color (eg 'monofast') to engage the fast inky.

ratmandu, 2019-10-27:
I set up a new color name ('fastAndFurious') for the waveshare fast update that I just put in, #428. Maybe going that route would work?

dipsylala, 2019-10-27:
Nice :D We might get a few folks suddenly complaining about the change in Inky display rate, but having a specific color makes the fast refresh 'opt-in'

coderobe, 2019-10-27:
Here's an example of the burn-in after <24h of running the pwnagotchi - in the gif, the pwnagotchi service is stopped and the screen is cleared to yellow using the proper method. Artifacts from the pwnagotchi screen are clearly visible.
[image]

I have been running a clear to yellow/black/white in a loop for about 30 minutes and the burn-in is lessening, but it is still not completely gone.