evilsocket/pwnagotchi #447: Systemctl restart pwnagotchi crashes UI modules [BUG]

PeterAlfonsLoch, 2019-10-27:
**Describe the bug**
Systemctl restart pwnagotchi crashes UI modules, So the Ui just stops working, and the process "pwnagotchi" needs to be killed and started, or the whole device restarted. The rest runs in the background as I saw in the logs, yet I saw no errors about the UI.

**To Reproduce**
Steps to reproduce the behavior:
1. Using command "systemctl restart pwnagotchi"

**Expected behavior**
Used to restart application pwnagotchi, recomendend when forced into manual mode to get back to automode

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.1.0RC0
 - OS version: Same as used by 1.1.0RC0
 - Type of hardware: Pi ZeroW + Inkyphat
 - Any additional hardware used: 5Ah Powerbank on PWR-only usb micro port

evilsocket, 2019-10-28:
please attach logs

closed 2019-10-28 by evilsocket

Declan0811, 2021-02-03:
> **Describe the bug**
> "systemctl restart pwnagotchi" crashes UI modules, So the UI just stops working, and the process "pwnagotchi" needs to be killed and started, or the whole device restarted. The rest runs in the background as I saw in the logs, yet I saw no errors about the UI.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. Using command "systemctl restart pwnagotchi"
> 
> **Expected behavior**
> Used to restart application pwnagotchi, recomendend when forced into manual mode to get back to automode
> 
> **Environment (please complete the following information):**
> 
> * Pwnagotchi version: 1.1.0RC0
> * OS version: Same as used by 1.1.0RC0
> * Type of hardware: Pi ZeroW + InkyPhat B/W
> * Any additional hardware used: 5Ah Powerbank on PWR-only usb micro port
> * User-level: root

I also have this issue :/
I don't know how to attach logs though