evilsocket/pwnagotchi #448: Refactor Waveshare 1.54 config option value

chillinPanda, 2019-10-27:
In #434 support for the screen was added.

**Is your feature request related to a problem? Please describe.**
It's not a problem now but could be in the future if someone wants to add new types of the display module and must create new config values which might confuse users.

At the moment the value is `waveshare154inch` but this is for the type B.

What happens if someone else adds support for type C? Then we would have

* `waveshare154inch` -> Type B
* for example `waveshare154inch_c` -> Type C

As of the vendor's website there are three types: https://www.waveshare.com/wiki/1.54inch_e-Paper_Module_(B)

* 1.54inch e-Paper Module
  * this has two versions, see their git: [epd1in54.py](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/lib/waveshare_epd/epd1in54.py) vs. [epd1in54_V2.py](https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/lib/waveshare_epd/epd1in54_V2.py)
* 1.54inch e-Paper Module (B)
* 1.54inch e-Paper Module (C)

**Describe the solution you'd like**

Rename option value to `waveshare154inch_b`

**Describe alternatives you've considered**
None.

**Additional context**

* Since it is not released yet, changing the option's value should not be a breaking change, or should it?
* Recommendations for the other type's names
  * `waveshare154inch_v1`
  * `waveshare154inch_v2`
  * `waveshare154inch_b`
  * `waveshare154inch_c`

chillinPanda, 2019-10-27:
Hi @ronangaillard ,

what do you think about the option's name?

ronangaillard, 2019-10-27:
Perfect, I should have done that myself first
I can do a PR tomorrow :)

ronangaillard, 2019-10-27:
I'll also update the font sizes to get a better reading on such screens