evilsocket/pwnagotchi #451: Pwnagotchi Birthday celebrations based on epochs

dipsylala, 2019-10-28:
Purely a fun thing to celebrate when an AI reaches a significant number of epochs. We can't really measure time, so an epoch 'feels' like a more suitable measure of age.

When an pwnagotchi  reaches a multiple of a significant number of epochs (say 1000) it's happy face says the following for the next few epochs when it has relatively neutral feelings (so that the user can still see the pwnagotchi is still working on its birthday)

`(*∇*)由  Happy birthday to me! I'm xxxx epochs old`

The face is made smaller (3 figures instead of 4)  to accommodate the cake so should still fit.

This has the benefit of reminding the owner that their pwnagotchis age and are always learning :)

Arttumiro, 2019-10-28:
Also to add to this, birthdays based on day of first boot (if possible that is)
But then celebrations on epoch milestones

deveth0, 2019-10-29:
Day of first boot might be a bit hard as the device does not not the time if you do not pair it over BT or share a internet connection.
But I like the birthday idea, might also be a party events to nearby pals?

leon-th, 2019-10-29:
The AI has an created on date in the brain.json

pingywon, 2019-11-20:
I would love to see "Total Epochs" as a plugin for the Waveshare v2