evilsocket/pwnagotchi #455: [BUG] backup.sh fails if zip isn't installed

ciara1234, 2019-10-28:
**Describe the bug**
Backup.sh fails if zip isn't installed. It appears zip doesn't come as default package in Ubuntu WSL.

Created PR https://github.com/evilsocket/pwnagotchi/pull/454 for this

**To Reproduce**
run ./backup.sh

**Expected behavior**
backup.sh should output a zip file

**Screenshots**
n/a

**Environment (please complete the following information):**
 - Pwnagotchi version : 1.0.1
 - OS version: Just using Ubuntu WSL in Windows 10
 - Type of hardware
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

luclu7, 2019-10-28:
Why not just use tar in fallback if zip isn't installed? It should be installed on basically every unix system.

ciara1234, 2019-10-28:
> Why not just use tar in fallback if zip isn't installed? It should be installed on basically every unix system.

tar won't have compression though - what's more important, getting a backup or having a compressed backup?

ciara1234, 2019-10-29:
Merged

closed 2019-10-29 by ciara1234