evilsocket/pwnagotchi #458: gpio buttons

lazee486, 2019-10-29:
original Tamagotchi had buttons to interact, I'd like to be able to interact more and add features.

using gpio for buttons to run scripts - 
shutdown
sleep (low power eg screen off but still working or slow refresh) 
run a specific attack like pixie dust.
proximity sensor screen off
view currently stored messages on screen

lexusburn, 2019-10-29:
There's already a GPIO plugin. Have a look at https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/defaults.yml ;-)

lazee486, 2019-10-30:
found it after i posted, thanks for not yelling rtfm lol, ive been reading
nonstop and making progress:)

akhepcat, 2019-11-07:
FYI,  https://www.instructables.com/id/Simple-Raspberry-Pi-Shutdown-Button/  works great; 
  header pin 5,  in the script use "gpio_pin_number=3"  and 4 seconds of push will shut the system down gracefully.  if you don't remove power,  pushing the button again will restart the system.

Now all we need is a "shutdown" image for the ePaper displays so that we know it's officially on the shutdown path.

xenDE, 2019-11-20:
@lazee486 can you close this issue?

scrypthekid, 2023-03-19:
Total novice here. So if I got this right I would have to edit this part of the gpio_buttons.py file like this:

` def on_loaded(self):
        logging.info("GPIO Button plugin loaded.")

        # get list of GPIOs and associated commands
        gpios = {
            X: "command1",    # X = GPIO pin number, "command" = command for the button press
            X: "command2",
            X: "command3",
            X: "command4"
        }`
and then set the "main.plugins.gpio_buttons.enabled = true" in the config.toml file or is it just to simple to be true?