evilsocket/pwnagotchi #459: bluetooth keyboard access messaging with display

lazee486, 2019-10-29:
be able to pair bluetooth keyboard and use display for terminal or at least as private messaging app. Know this is a big feature request

Arttumiro, 2019-10-29:
You can ssh in and use it through your phone with bt tether
Thats what i do

lazee486, 2019-10-29:
Thanks, im still building my first one, keeps failing so havent tested properly :)

dadav, 2019-10-30:
It's planed to create a chat-web-plugin, as soon as we have a good solution for the CORS problem. So you can chat via the browser of your phone. Does that sound good to you? :)

lazee486, 2019-10-30:
sounds great :) i got my unit up ( no display yet) and im loving it :)

OldFolksdonthack, 2021-01-01:
I have a display but cannot SSH in nor see in when connected to my box. flipside plugged in hdmi and got what in the photo. wish I hade a keyboard 
[image]