evilsocket/pwnagotchi #460: [BUG] Always in MANU mode on Pi3

maxxer, 2019-10-29:
**Describe the bug**
1.1.0RC0 on Pi3 always starts in MANU mode, despite of the ethernet cable being connected or not

**Expected behavior**
When ethernet cable is disconnected it should start in AUTO mode.

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.1.0RC0
 - Type of hardware: Raspberry Pi3+

**Additional context**
A temporary workaround has been posted [on Reddit](https://www.reddit.com/r/pwnagotchi/comments/dnz5p2/pi3_always_boots_to_manu/f5j3up9/), that is tweaking the `eth0` check in `/usr/bin/pwnagotchi-launcher` by making it return _false_.

Nodeswitch, 2019-10-29:
Can be replicated on the Pi 4, also.

maxxer, 2019-10-29:
Fix merged

closed 2019-10-29 by maxxer