evilsocket/pwnagotchi #463: Don't spend time on "useless" APs

coderobe, 2019-10-29:
**Is your feature request related to a problem? Please describe.**
There are a few networks that are never worth trying to pwn, like open hotspots with login walls, or user-based enterprise networks - right now, pwnagotchi will still spend valuable time associating with these networks and collecting handshakes.

**Describe the solution you'd like**
- Don't associate with open networks
- Ignore handshakes from EAP / enterprise networks

**Describe alternatives you've considered**
- Ship a default whitelist of networks i.e.
  - eduroam - user based auth (EAP)
  - Telekom_FON - open hotspot with login wall
  - ...

**Additional context**
I'm not sure if it makes sense to collect handshakes from enterprise networks, as those are usually certificate based, user+pass, or use different means of login a la EAP

darrenmartyn, 2019-10-30:
Yep, noticed this as well with the Vodafone Homespot and Telekom ones in Germany. They are "open", but it spends time on them. Maybe a filtering mechanism...

evilsocket, 2019-10-30:
this is a bug bettercap's side, it shouldn't happen ... you guys sure those are open aps?

evilsocket, 2019-10-30:
fixed with https://github.com/evilsocket/pwnagotchi/commit/1600d8cbd14a0d3343c3d925da86b99d5a6ac296

closed 2019-10-30 by evilsocket

coderobe, 2019-10-30:
what's your opinion on the second half of this issue RE: networks like `eduroam` which use user:pass auth or certificates, can these be filtered too or is there not enough information available at this stage to be able to filter these automatically?

evilsocket, 2019-10-30:
i'd say the latter

commit 1600d8cb (2019-10-30): fix: skipping open access points (fixes #463)