evilsocket/pwnagotchi #464: Improve feedback on configuration errors

dipsylala, 2019-10-29:
It's mainly an issue with new users to the system experiencing the configuration structure and making mistakes.

When an invalid config.yml file is parsed and an error occurs, a full traceback is generated. To new users, it's not obvious what's happened. Most of the traceback is irrelevant, and only the first couple of lines are needed.

**Suggested solution**
An explicit notification that there was an error in the user configuration file, with the key details, capturing the YML ParseError

dadav, 2019-10-30:
good idea, more accurate exception handling is always better^^

dipsylala, 2019-10-30:
The PR associated with this got merged in, so I'm happy to close

closed 2019-10-30 by dipsylala