evilsocket/pwnagotchi #466: Invalid Slack Link

flamespinner, 2019-10-30:
The Slack link in the Readme.md is serving an Application Error when browsed to (https://pwnagotchi.herokuapp.com/). 

> Application error
An error occurred in the application and your page could not be served. If you are the application owner, check your logs for details. You can do this from the Heroku CLI with the command
heroku logs --tail

hexwaxwing, 2019-10-30:
it will work again in 24 hours. in the meantime, please use https://join.slack.com/t/pwnagotchi/shared_invite/enQtODE2NTY4MTYwOTUwLTAzNzRiZjI0ZWQ5Yzg3ODAzMzJhNGQ3NThmZWIxZmJlZTliMzFiMzAyNTI4MTY3OWQwOWY3ZjE5Y2ViMzYxMTY .

evilsocket, 2019-10-30:
@hexwaxwing can you fix it or is it better if i just replace the form with something else?

hexwaxwing, 2019-10-30:
**TL;DR: this is a one time problem that won't ever happen again; please just use the temporary invite link for the next 24 hours. i don't think it's worth the trouble of reimplementing differently because i will ensure this doesn't occur again in the future.**

longer explanation:
heroku's free tier allocates 550 dyno hours for an app per month, and we passed that monthly quota at 1:46 AM CEST today.

heroku will allow us to allocate an additional 450 dyno hours per month (for a total of 1000 dyno hours per month, instead of the default 550) once i verify the account with a card, but i cannot do that until i get my paycheck (which will be tomorrow, oct 31). so, the app will continue to error out until i verify the heroku account with a card (or, worst case scenario, nov 1 happens). but i'll verify the account as soon as i get paid tomorrow and then it'll be fully functional from that point on.

this won't happen again. i'm sorry about this happening this month. am broke.

evilsocket, 2019-10-30:
replaced with https://invite.pwnagotchi.ai/ ... thanks @hexwaxwing you can deactivate the heroku app now.