evilsocket/pwnagotchi #468: Give it a "home network"

kimusan, 2019-10-30:
**Is your feature request related to a problem? Please describe.**
Personally I dont want to tether my pwnagotchi via my mobile phone so I was hoping that we could get a way of letting an pwnagotchi know when it is home and connect to the wifi there.

**Describe the solution you'd like**
Since the pwnagotchi supports wifi it should be pretty trivial to tell it that bssid XXXX is its home network and then let it connect and upload whenever it gets near that network.  if needed, it could change to MANU mode in the same manner as if it gets the usb data connection. 
This would also make it possible to let it use wifi tethering on the mobile phone which is a lot easier to configure on the phone compared to bt-pan.

cd-rum, 2019-10-30:
This is already covered in #207

kimusan, 2019-10-30:
thanks did not notice that one. looking forward for that feature.

closed 2019-10-30 by kimusan