evilsocket/pwnagotchi #474: Request adding support for the Waveshare 2.9 inch e-ink display

opteeks, 2019-10-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add the drivers required to support the Waveshare 2.9 inch e-ink display by Waveshare
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

coderobe, 2019-10-31:
Should work (https://github.com/evilsocket/pwnagotchi/issues/360)

opteeks, 2019-10-31:
I just submitted a PR to add the drivers for the 2.9 inch display. I've tested it thoroughly with my pi zero w and display and have the layout finalized.

evilsocket, 2019-10-31:
@opteeks your PR has conflicts, i need you to fix them before i can merge

opteeks, 2019-11-03:
Conflicts have been resolved.

closed 2019-11-03 by evilsocket

TheSidd, 2020-06-20:
@opteeks has this been resolved? What do I need to add in config to make it work?

opteeks, 2020-06-20:
This was already resolved and has been working fine.