evilsocket/pwnagotchi #480: [BUG] Always in MANU mode on PiZ

deveth0, 2019-10-31:
**Describe the bug**
Related to the fix from #460: now it seems to work with a Ri3/4 but the PiZ starts in MANU mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 1.1.0 on PiZ
2. Power it, start and wait

**Expected behavior**
Should start in AUTO but starts in MANU

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.1.0
 - Type of hardware: PiZ

evilsocket, 2019-10-31:
fixed by https://github.com/evilsocket/pwnagotchi/pull/461 ... will be released soon as 1.1.1

closed 2019-10-31 by evilsocket

deveth0, 2019-10-31:
Are you sure? As far as I can see #461 is already included in 1.1.0:
https://github.com/evilsocket/pwnagotchi/blob/v1.1.0/builder/pwnagotchi.yml#L307

When removing the part `|| [[ !$(grep '1' /sys/class/net/eth0/carrier) ]]` it works as expected.

evilsocket, 2019-10-31:
i'm refactoring this as we speak