evilsocket/pwnagotchi #486: Angry face for Pwnagotchi

mrseeker, 2019-11-01:
**feature request**
While I was discussing the Pwnagotchi with someone else, we came up with the idea of having an "angry face", that can be used when something occurs that causes the Pwnagotchi to miss a handshake. This is similar to the "bored" and "crying" face, but with small difference, eg:
1. The handshake is partial/incomplete
2. The deauth does not work
3. No new handshakes, despite discovering different APs (similar to bored)
4. Not being fed information (eg WiFi malfunctions, or running it on a Pi without WiFi capabilities (pi zero without Wireless)
5. Not hitting his target after an epoch
6. General error messages
7. When trying a more aggressive strategy.

In general, it is a face that can be seen when the Pwnagotchi is struggling or annoyed that it cannot reach its ultimate goal, or feels some way to express anger at its owner.

evilsocket, 2019-11-01:
mmm from what i'm reading this face should basically replace the bored and missed states?

mrseeker, 2019-11-01:
Not really replace it, but more as an act of "frustration" for ignoring the signals, or when it goes on a failure streak (eg keeps losing points). Sad and bored should be prioritized over anger, but if you start crying and get ignored, you eventually start to feel frustrated. That is the mood I was thinking of: at first it starts being bored, and after an X amount of time it should become angry because you keep ignoring it.
Think of a cat: it first starts meowing for food, and when you keep ignoring it will eventually develop a very grumpy face and start doing mischief...

evilsocket, 2019-11-01:
the current transition (assuming that the number of inactive epochs keeps increasing) is:

    normal -> bored -> sad

we could do:

    normal -> bored -> sad -> angry

(i understood what you meant but this is code, not a cat, so i need to translate that to state transitions :D)

commit 31a89cbe (2019-11-01): new: added new angry state (closes #486)
closed 2019-11-01 by evilsocket