evilsocket/pwnagotchi #491: Feature Request: Allow Whitelist by the vendor part of MAC address

DaniloNC, 2019-11-01:
Some times the list of BSSIDs to whitelist is so big that it might be easier to just whitelist by the vendor part of the mac address. 

It would be good if the grid plugin and the whitelist option would accept filtering based only on the vendor part of mac address, for instance, suppose a user wants to whitelist all network whose vendor is `FA:8F:CA`, a configuration to ignore this could possibly be: 

```yaml
main:
    plugins:
        grid:
            enabled: true
            report: false # don't report pwned networks by default!
            exclude: # do not report the following networks (accepts both ESSIDs and BSSIDs)
                - FA:8F:CA

    # access points to ignore
    whitelist:
        - FA:8F:CA
```

dadav, 2019-11-02:
hm, this needs some bettercap modification, right?

evilsocket, 2019-11-02:
nope those filters are applied python side

evilsocket, 2019-11-04:
@DaniloNC i've checked the code and it turns out you can already do that! you can use, in the config, `main.filter` that if defined, it'll be a regexp that will be executed against all access points: if they match, they'll be included in the list, otherwise they'll be skipped.

For the grid plugin instead, the exclude parameter already supports partial matches 👍

closed 2019-11-04 by evilsocket

DaniloNC, 2019-11-08:
Thank you for the reply @evilsocket!

I think what I want to do is still not possible using filters, in my case I want to accept everything but the mac vendor `00:FF:00:xx:xx:xx`, reading the code at https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/agent.py#L159-L162 , the filter logic is to filter what networks to **include**, and not what networks to **exclude**.

```python
    def _filter_included(self, ap):
        return self._filter is None or \
               self._filter.match(ap['hostname']) is not None or \
               self._filter.match(ap['mac']) is not None
```

If I use filter parameter with the following syntax, I would be  **allowing** attacks to vendor mac address starting with `00:ff:00`, but what I want is to **deny** attacks to  vendor mac address `00:ff:00`.

```yaml
main:
    whitelist:
        - EXAMPLE_NETWORK
    # if not null, filter access points by this regular expression
    filter: ^00:ff:00:.*
```

It should be possible to create a filter value that matches everything but 00:ff:00, but that is not user-friendly. Furthermore, the agent logic is to check if the regex matches either the mac or ssid, creating a filter that matches everything but the vendor mac, would probably match any ssid, making this approach difficult or impossible to create.

Is my understanding correct? Maybe I am missing something.
How should I configure pwnagotchi to not attack any networks that the mac vendor starts with `00:ff:00` ?

DaniloNC, 2019-11-08:
A possible solution would be https://github.com/DaniloNC/pwnagotchi/blob/0bfcba14eff5bb01c444868d051ee8f25577a9af/pwnagotchi/agent.py#L179

If that is okay, I can create a pull request.