evilsocket/pwnagotchi #494: random bluetoothd crashes

evilsocket, 2019-11-01:
seems related

```
[ 2903.917746] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.917753] sysfs group 'byte_queue_limits' not found for kobject 'tx-0'
[ 2903.917757] Modules linked in: evdev uinput bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil raspberrypi_hwmon snd_bcm2835(C) hwmon cfg80211 snd_pcm snd_timer rfkill snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835aux v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 2903.917930] CPU: 0 PID: 4695 Comm: kbnepd bnep0 Tainted: G         C        4.19.66-Re4son+ #1
[ 2903.917935] Hardware name: BCM2835
[ 2903.917969] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 2903.917992] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 2903.918075] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 2903.918092] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 2903.918112] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 2903.918131] [<c021f370>] (sysfs_remove_group) from [<c0613460>] (netdev_queue_update_kobjects+0xc0/0x178)
[ 2903.918146] [<c0613460>] (netdev_queue_update_kobjects) from [<c0613578>] (netdev_unregister_kobject+0x60/0x80)
[ 2903.918169] [<c0613578>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 2903.918185] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 2903.918199] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 2903.918237] [<c05eb568>] (unregister_netdev) from [<bf5f1958>] (bnep_session+0x4d4/0x888 [bnep])
[ 2903.918288] [<bf5f1958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 2903.918305] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 2903.918312] Exception stack(0xd67affb0 to 0xd67afff8)
[ 2903.918321] ffa0:                                     00000000 00000000 00000000 00000000
[ 2903.918332] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2903.918340] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2903.918347] ---[ end trace cfb2c6995ec3c276 ]---
...
[ 2903.918482] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.918488] sysfs group 'power' not found for kobject 'bnep0'
[ 2903.918492] Modules linked in: evdev uinput bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil raspberrypi_hwmon snd_bcm2835(C) hwmon cfg80211 snd_pcm snd_timer rfkill snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835aux v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 2903.918652] CPU: 0 PID: 4695 Comm: kbnepd bnep0 Tainted: G        WC        4.19.66-Re4son+ #1
[ 2903.918657] Hardware name: BCM2835
[ 2903.918682] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 2903.918701] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 2903.918722] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 2903.918738] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 2903.918756] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 2903.918781] [<c021f370>] (sysfs_remove_group) from [<c04a1474>] (dpm_sysfs_remove+0x58/0x60)
[ 2903.918798] [<c04a1474>] (dpm_sysfs_remove) from [<c0494e48>] (device_del+0x7c/0x360)
[ 2903.918813] [<c0494e48>] (device_del) from [<c0613594>] (netdev_unregister_kobject+0x7c/0x80)
[ 2903.918832] [<c0613594>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 2903.918847] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 2903.918861] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 2903.918891] [<c05eb568>] (unregister_netdev) from [<bf5f1958>] (bnep_session+0x4d4/0x888 [bnep])
[ 2903.918919] [<bf5f1958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 2903.918933] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 2903.918940] Exception stack(0xd67affb0 to 0xd67afff8)
[ 2903.918949] ffa0:                                     00000000 00000000 00000000 00000000
[ 2903.918959] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2903.918968] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2903.918973] ---[ end trace cfb2c6995ec3c277 ]---
...
[ 2903.919015] WARNING: CPU: 0 PID: 4695 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 2903.919021] sysfs group 'statistics' not found for kobject 'bnep0'
[ 2903.919024] Modules linked in: evdev uinput bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil raspberrypi_hwmon snd_bcm2835(C) hwmon cfg80211 snd_pcm snd_timer rfkill snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) i2c_bcm2835 bcm2835_mmal_vchiq(C) spi_bcm2835aux v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 2903.919178] CPU: 0 PID: 4695 Comm: kbnepd bnep0 Tainted: G        WC        4.19.66-Re4son+ #1
[ 2903.919182] Hardware name: BCM2835
[ 2903.919201] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 2903.919219] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 2903.919237] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 2903.919252] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 2903.919269] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 2903.919284] [<c021f370>] (sysfs_remove_group) from [<c021f3b0>] (sysfs_remove_groups+0x34/0x44)
[ 2903.919310] [<c021f3b0>] (sysfs_remove_groups) from [<c0492a40>] (device_remove_attrs+0x4c/0x7c)
[ 2903.919326] [<c0492a40>] (device_remove_attrs) from [<c0494f14>] (device_del+0x148/0x360)
[ 2903.919339] [<c0494f14>] (device_del) from [<c0613594>] (netdev_unregister_kobject+0x7c/0x80)
[ 2903.919357] [<c0613594>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 2903.919372] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 2903.919386] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 2903.919414] [<c05eb568>] (unregister_netdev) from [<bf5f1958>] (bnep_session+0x4d4/0x888 [bnep])
[... 129 lines cut ...]
[ 3631.199926] [<c0065d9c>] (vprintk_store) from [<c0066244>] (vprintk_emit+0x9c/0x340)
[ 3631.199930] [<c0066244>] (vprintk_emit) from [<c00666c8>] (vprintk_default+0x40/0x74)
[ 3631.199934] [<c00666c8>] (vprintk_default) from [<c006716c>] (vprintk_func+0x48/0xb0)
[ 3631.199938] [<c006716c>] (vprintk_func) from [<c0063df4>] (vprintk+0x18/0x1c)
[ 3631.199941] [<c0063df4>] (vprintk) from [<c00245b0>] (__warn+0x7c/0x118)
[ 3631.199945] [<c00245b0>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 3631.199949] [<c002469c>] (warn_slowpath_fmt) from [<c06f7c74>] (kobject_get+0x50/0x5c)
[ 3631.199953] [<c06f7c74>] (kobject_get) from [<c049438c>] (device_add+0xc0/0x608)
[ 3631.199957] [<c049438c>] (device_add) from [<c0613614>] (netdev_register_kobject+0x7c/0x160)
[ 3631.199961] [<c0613614>] (netdev_register_kobject) from [<c05ebb8c>] (register_netdevice+0x390/0x568)
[ 3631.199966] [<c05ebb8c>] (register_netdevice) from [<c05ebd90>] (register_netdev+0x2c/0x44)
[ 3631.199970] [<c05ebd90>] (register_netdev) from [<bf5f1f20>] (bnep_add_connection+0x214/0x30c [bnep])
[... cut ...]
[ 3631.199994] [<c01ae678>] (sys_ioctl) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[ 3631.199997] Exception stack(0xd8b27fa8 to 0xd8b27ff0)
[ 3631.200001] 7fa0:                   00000000 000f1828 0000000a 400442c8 bec1e458 00000001
[ 3631.200005] 7fc0: 00000000 000f1828 002af556 00000036 00000014 00000000 b6e59448 00000000
[ 3631.200008] 7fe0: 000f1c5c bec1e454 0003ba58 b6bb951c
[ 3631.200011] Code: 3a000019 e3530000 e243e001 0a00001b (e5d2c000) 
[ 3631.200014] ---[ end trace cfb2c6995ec3c27b ]---
```

evilsocket, 2019-11-01:
@dadav fyi

evilsocket, 2019-11-01:
it might be related to #267 as they appeared together in my case

dadav, 2019-11-02:
https://github.com/evilsocket/pwnagotchi/issues/377#issuecomment-546714710

he got the same error...

akhepcat, 2019-11-08:
seeing this on the latest release.  means i can't use BNEP for network access at all

neurovish, 2019-11-22:
Pasting the details here I mentioned in #267. So far, the crashes don't seem random for me.

I get the "brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4109, -110" errors after about an hour when I have bluetooth enabled in the config, but bluetooth does not connect for whatever reason. pi0w, v1.3.0, temp shows as 41c. pwnagotchi process locks and goes defunct when this happens. Is this the same bug?

Bluetooth enabled and connected: No problems after 2 hours
Bluetooth disabled: No problems after 2 hours