evilsocket/pwnagotchi #495: switch to a functionally equivalent but less space-centric alternative to yaml config

evilsocket, 2019-11-01:
as per title, we need to switch to a functionally equivalent but less space-centric alternative to yaml config, specifically, something that breaks less easily when tabs and spaces are mixed. it can be something as stupid as:

section.subsection.fuckyaml "blablabla"
section.subsection.fuckyaml.harder true

or still indented by less idiotic than yaml

dadav, 2019-11-01:
hm...maybe TOML?

```toml
[main]
name = ""
lang = "en"
iface = "mon0"
mon_start_cmd = "/usr/bin/monstart"
mon_stop_cmd = "/usr/bin/monstop"
mon_max_blind_epochs = 50.0
log = "/var/log/pwnagotchi.log"
no_restart = false
whitelist = [
  "EXAMPLE_NETWORK",
  "ANOTHER_EXAMPLE_NETWORK"
]

  [main.plugins]

    [main.plugins.grid]
    enabled = true
    report = false
    exclude = [
      "YourHomeNetworkHere"
    ]

    [main.plugins.auto-update]
    enabled = true
    install = true
    interval = 1.0

    [main.plugins.auto-backup]
    enabled = false
    interval = 1.0
    files = [
      "/root/brain.nn",
      "/root/brain.json",
      "/root/.api-report.json",
      "/root/handshakes/",
      "/root/peers/",
      "/etc/pwnagotchi/",
      "/var/log/pwnagotchi.log"
    ]
    commands = [
      "tar czf /root/pwnagotchi-backup.tar.gz {files}"
    ]

    [main.plugins.net-pos]
    enabled = false
    api_key = "test"

    [main.plugins.gps]
    enabled = false
    speed = 19200.0
    device = "/dev/ttyUSB0"

    [main.plugins.twitter]
    enabled = false
    consumer_key = "aaa"
    consumer_secret = "aaa"
    access_token_key = "aaa"
    access_token_secret = "aaa"

    [main.plugins.onlinehashcrack]
    enabled = false

    [main.plugins.wpa-sec]
    enabled = false
    api_url = "https://wpa-sec.stanev.org"

    [main.plugins.wigle]
    enabled = false

    [main.plugins.screen_refresh]
    enabled = false
    refresh_interval = 50.0

    [main.plugins.quickdic]
    enabled = false
    wordlist_folder = "/opt/wordlists/"

    [main.plugins.AircrackOnly]
    enabled = false

    [main.plugins.bt-tether]
    enabled = false

      [main.plugins.bt-tether.devices]

        [main.plugins.bt-tether.devices.android-phone]
        enabled = false
        search_order = 1.0
        ip = "192.168.44.44"
        netmask = 24.0
        interval = 1.0
        scantime = 10.0
        max_tries = 10.0
        share_internet = false
        priority = 1.0

        [main.plugins.bt-tether.devices.ios-phone]
        enabled = false
        search_order = 2.0
        ip = "172.20.10.6"
        netmask = 24.0
        interval = 5.0
        scantime = 20.0
        max_tries = 0.0
        share_internet = false
        priority = 999.0

    [main.plugins.memtemp]
    enabled = false
    orientation = "horizontal"

    [main.plugins.pawgps]
    enabled = false
    ip = ""

    [main.plugins.gpio_buttons]
    enabled = false

      [[main.plugins.gpio_buttons.gpios]]
      20 = "sudo touch /root/.pwnagotchi-auto && sudo systemctl restart pwnagotchi"

      [[main.plugins.gpio_buttons.gpios]]
      21 = "shutdown -h now"

[ai]
enabled = true
path = "/root/brain.nn"
laziness = 0.1
epochs_per_episode = 50.0

  [ai.params]
  gamma = 0.99
  n_steps = 1.0
  vf_coef = 0.25
  ent_coef = 0.01
  max_grad_norm = 0.5
  learning_rate = 0.001
  alpha = 0.99
  epsilon = 0.00001
  verbose = 1.0
  lr_schedule = "constant"

[personality]
advertise = true
deauth = true
associate = true
channels = []
min_rssi = -200.0
ap_ttl = 120.0
sta_ttl = 300.0
recon_time = 30.0
max_inactive_scale = 2.0
recon_inactive_multiplier = 2.0
hop_recon_time = 10.0
min_recon_time = 5.0
max_interactions = 3.0
max_misses_for_recon = 5.0
excited_num_epochs = 10.0
bored_num_epochs = 15.0
sad_num_epochs = 25.0
bond_encounters_factor = 20000.0

[ui]
fps = 0.0

  [ui.faces]
  look_r = "( ⚆_⚆)"
  look_l = "(☉_☉ )"
  look_r_happy = "( ◕‿◕)"
  look_l_happy = "(◕‿◕ )"
  sleep = "(⇀‿‿↼)"
  sleep2 = "(≖‿‿≖)"
  awake = "(◕‿‿◕)"
  bored = "(-__-)"
  intense = "(°▃▃°)"
  cool = "(⌐■_■)"
  happy = "(•‿‿•)"
  excited = "(ᵔ◡◡ᵔ)"
  grateful = "(^‿‿^)"
  motivated = "(☼‿‿☼)"
  demotivated = "(≖__≖)"
  smart = "(✜‿‿✜)"
  lonely = "(ب__ب)"
  sad = "(╥☁╥ )"
  angry = "(-_-')"
  friend = "(♥‿‿♥)"
  broken = "(☓‿‿☓)"
  debug = "(#__#)"

  [ui.display]
  enabled = true
  rotation = 180.0
  type = "waveshare_2"
  color = "black"

    [ui.display.video]
    enabled = true
    address = "0.0.0.0"
    port = 8080.0
    on_frame = ""

[bettercap]
scheme = "http"
hostname = "localhost"
port = 8081.0
username = "pwnagotchi"
password = "pwnagotchi"
handshakes = "/root/handshakes"
silence = [
  "ble.device.new",
  "ble.device.lost",
  "ble.device.disconnected",
  "ble.device.connected",
  "ble.device.service.discovered",
  "ble.device.characteristic.discovered",
  "wifi.client.new",
  "wifi.client.lost",
  "wifi.client.probe",
  "wifi.ap.new",
  "wifi.ap.lost",
  "mod.started"
]
```

dadav, 2019-11-01:
You can even omit the `[sections]` and do:

```toml
main.name = ""
main.lang = "en"
main.iface = "mon0"
main.mon_start_cmd = "/usr/bin/monstart"
main.mon_stop_cmd = "/usr/bin/monstop"
main.mon_max_blind_epochs = 50.0
main.log = "/var/log/pwnagotchi.log"
main.no_restart = false
```

and so on...

dadav, 2019-11-02:
I suggest using the `tomlkit` module. It preserves the style of the configuration.

evilsocket, 2019-11-02:
what about just ini files? https://docs.python.org/3/library/configparser.html

dadav, 2019-11-02:
> what about just ini files? https://docs.python.org/3/library/configparser.html

I'm not a big fan of ini's :D But that aside, what is don't like about this configparser:
- every value is a string (we have to convert the values by ourselfs)
- only one level possible; no subsections (can't do [main][plugins][grid])
- spaces in keys are allowed (wtf)

evilsocket, 2019-11-02:
conversation https://community.pwnagotchi.ai/t/configuration-format-change/35/3

dadav, 2020-01-19:
Derek close

closed 2020-01-19 by derek[bot]