evilsocket/pwnagotchi #503: [BUG] skript linux_connection_share.sh don't work

daniel156161, 2019-11-02:
i updated today to version 1.1.1 and now the linux_connection_share.sh don't works any more can any one help me?

 - Pwnagotchi version: 1.1.1
 - OS version: Linux Ubuntu 5.3.0-19-generic #20-Ubuntu SMP Fri Oct 18 09:04:39 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

editet the interfaces to them on my computer

#!/usr/bin/env bash

# name of the ethernet gadget interface on the host
USB_IFACE=${1:-enp0s20f0u2i1}
USB_IFACE_IP=10.0.0.1
USB_IFACE_NET=10.0.0.0/24
# host interface to use for upstream connection
UPSTREAM_IFACE=${2:-wlp2s0}

#ip addr add "$USB_IFACE_IP/24" dev "$USB_IFACE"
#ip link set "$USB_IFACE"

iptables -A FORWARD -o "$UPSTREAM_IFACE" -i "$USB_IFACE" -s "$USB_IFACE_NET" -m conntrack --ctstate NEW -j ACCEPT
iptables -A FORWARD -m conntrack --ctstate ESTABLISHED,RELATED -j ACCEPT
iptables -t nat -F POSTROUTING
iptables -t nat -A POSTROUTING -o "$UPSTREAM_IFACE" -j MASQUERADE

echo 1 > /proc/sys/net/ipv4/ip_forward

daniel156161, 2019-11-02:
oh the raspberry use a other dns as my network and i have blocked a other dns as my one

closed 2019-11-02 by daniel156161

hackabean, 2019-11-03:
@daniel156161 how did you manage to fix it, can you provide more detailed instructions ?

daniel156161, 2019-11-03:
a very simple fix i have blocked all other DNS Server not that into my pfSense VM

i added it the Firewall a new roles that my pc that i using with my Pwnagotchi can go to all DNS my Computer like

i has got to https://ip/firewall_rules.php?if=lan and added above my Block all other DNS a new role 

Protokoll: IPv4 UDP
Quelle: one Host or Alias IP of my PC
Ziel: all
Range of destination ports: from DNS(53) to DNS(53)

add a description to my new Role and saved it and now it works very fine