evilsocket/pwnagotchi #508: [BUG] bt-tether not connecting to an Android phone

hackabean, 2019-11-03:
Hello, thanks for an amazing project. I have been trying to set up bt-tethering option since yesterday. I have installed latest release and read on on the latest changes regarding bt-tether. For this I have used new config.yml that has been recently updated. My config.yml about bt-tether looks like this:
```
        bt-tether:
            enabled: true # if you want to use this, set ui.display.video.address to 0.0.0.0
            devices:
              android-phone:
                enabled: true
                search_order: 1 # search for this first
                mac: 'EC:D0:XX:XX:XX:XX' # mac of your bluetooth device
                ip: '192.168.44.44' # ip from which your pwnagotchi should be reachable
                netmask: 24
                interval: 1 # check every minute for device
                scantime: 10 # search for 10 seconds
                max_tries: 10 # after 10 tries of "not found"; don't try anymore
                share_internet: false
                priority: 1 # low routing priority; ios (prio: 999) would win here
              ios-phone:
                enabled: false
                search_order: 2 # search for this second
                mac: ~ # mac of your bluetooth device
                ip: '172.20.10.6' # ip from which your pwnagotchi should be reachable
                netmask: 24
                interval: 5 # check every 5 minutes for device
                scantime: 20
                max_tries: 0 # infinity
                share_internet: false
                priority: 999 # routing priority

```

I also replaced bt-tether.py with this updated one https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/bt-tether.py.

I have paired the phone and rpi0w manually using bluetoothctl, it said that the pairing was successful but it immediately disconnected shortly after. Same situation with an iphone as well.

What have I missed, is there anything else I can try ?

dadav, 2019-11-03:
Please run `pwnagotchi --debug` and post the logs :)

mrseeker, 2019-11-04:
Same issue here, running Android 9 (moto 7g). The phone does pair without issues but the phone does not recognize the pwnagotchi as a networked device. Can send debug when I get access.

hackabean, 2019-11-05:
@dadav 

Here are the logs from pwnagotchi --debug.

regarding:
[2019-11-05 11:57:21,708] [DEBUG] module 'pwnagotchi.plugins' has no attribute 'Plugin'

Since I have noticed the development regarding bt-tether i replaced:
```
root@pcap-shrimp:/# find / -name "bt-tether.py"
find: ‘/proc/3130’: No such file or directory
/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/bt-tether.py
/usr/local/src/pwnagotchi/build/lib/pwnagotchi/plugins/default/bt-tether.py
/usr/local/src/pwnagotchi/pwnagotchi/plugins/default/bt-tether.py
```
with https://raw.githubusercontent.com/evilsocket/pwnagotchi/master/pwnagotchi/plugins/default/bt-tether.py.

Not sure if I should have done that. Advice is welcomed.

```
root@pcap-shrimp:/# pwnagotchi --debug
[2019-11-05 11:57:21,705] [WARNING] error while loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/bt-tether.py: module 'pwnagotchi.plugins' has no attribute 'Plugin'
[2019-11-05 11:57:21,708] [DEBUG] module 'pwnagotchi.plugins' has no attribute 'Plugin'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 51, in load_from_path
    name, plugin = load_from_file(filename)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 43, in load_from_file
    spec.loader.exec_module(instance)
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/bt-tether.py", line 417, in <module>
    class BTTether(plugins.Plugin):
AttributeError: module 'pwnagotchi.plugins' has no attribute 'Plugin'
[2019-11-05 11:57:24,479] [INFO] grid plugin loaded.
[2019-11-05 11:57:24,482] [INFO] [update] plugin loaded.
[2019-11-05 11:57:24,494] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-11-05 11:57:24,498] [INFO] initializing waveshare v2 display
[2019-11-05 11:57:24,521] [INFO] web ui available at http://0.0.0.0:8080/
[2019-11-05 11:57:26,418] [INFO] pcap-shrimp@473d4f4d450a8ce75d87fe31a7768d1f6f114300f70ec93617ef9df2756da3cf (v1.1.1)
[2019-11-05 11:57:27,120] [DEBUG] effective configuration:

ai:
  enabled: true
  epochs_per_episode: 50
  laziness: 0.1
  params:
    alpha: 0.99
    ent_coef: 0.01
    epsilon: 1.0e-05
    gamma: 0.99
    learning_rate: 0.001
    lr_schedule: constant
    max_grad_norm: 0.5
    n_steps: 1
    tensorboard_log: null
    verbose: 1
    vf_coef: 0.25
  path: /root/brain.nn
bettercap:
  handshakes: /root/handshakes
  hostname: localhost
  password: pwnagotchi
  port: 8081
  scheme: http
  silence:
  - ble.device.new
  - ble.device.lost
  - ble.device.disconnected
  - ble.device.connected
  - ble.device.service.discovered
[... 171 lines cut ...]
  fps: 0.0

[2019-11-05 11:57:27,147] [DEBUG] plugin 'grid' v1.0.1 loaded from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2019-11-05 11:57:27,153] [DEBUG] plugin 'auto-update' v1.1.1 loaded from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2019-11-05 11:57:27,158] [INFO] entering auto mode ...
[2019-11-05 11:57:27,179] [INFO] [ai] bootstrapping dependencies ...
[2019-11-05 11:57:28,193] [INFO] connecting to http://localhost:8081/api ...
[2019-11-05 11:57:30,540] [INFO] found monitor interface: mon0
[2019-11-05 11:57:30,544] [INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]
[2019-11-05 11:57:30,555] [INFO] handshakes will be collected inside /root/handshakes
[2019-11-05 11:57:31,741] [DEBUG] restarting wifi module ...
[2019-11-05 11:57:35,096] [DEBUG] agent.next_epoch()
[2019-11-05 11:57:35,139] [INFO] [epoch 0] duration=00:00:08 slept_for=00:00:00 blind=0 inactive=1 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=33% mem=30% temperature=44C reward=-0.2
[2019-11-05 11:57:35,159] [DEBUG] RECON 30s
[2019-11-05 11:57:36,256] [DEBUG] polling events ...
[2019-11-05 11:57:39,048] [DEBUG] polling events ...
[2019-11-05 11:57:41,598] [DEBUG] polling events ...
[2019-11-05 11:57:44,398] [DEBUG] polling events ...
[2019-11-05 11:57:47,414] [DEBUG] polling events ...
[2019-11-05 11:57:50,205] [DEBUG] polling events ...
[2019-11-05 11:57:52,777] [DEBUG] polling events ...
[2019-11-05 11:57:54,877] [DEBUG] polling pwngrid-peer for peers ...
```

Problem I have noticed is that rpi has troubles connecting to the device (Android) when bluetooth-thethering is on. I have tried on two seperate Android phones and the situation was the same.

mrseeker, 2019-11-05:
@hackabean have you tried putting your bluetooth mac address between quotes? Pwnagotchi complains that the mac address could not be resolved. ([ERROR] BT-TETHER: Please specify the mac in your config.yml)

mac: 'EC:D0:9F:xx:xx:xx'

hackabean, 2019-11-05:
@mrseeker  Yup it is in single quotes. I am looking into my problem right now and it seems like my device is not creating bt-pan network adapter when I switch bluetooth-tethering ON in the settings, did you manage to get it working ?

yecukeng, 2019-11-05:
> settings, did you managed to get it working ?

Hi All! Looks like the recent changes about "refactoring the plugins to use
classes" are only exists in master but not in ver 1.1.1. So you have a "module
'pwnagotchi.plugins' has no attribute 'Plugin'" error with the new version
of bt-tether plugin. Also, wiki has been already updated to use the new
version of bt-there configuration, thats where "ERROR] BT-TETHER: Please
specify the mac in your config" came from..

mrseeker, 2019-11-05:
@yecukeng it seems you are right, the defaults.yml shows the old config...

https://github.com/evilsocket/pwnagotchi/blob/v1.1.1/pwnagotchi/defaults.yml

yecukeng, 2019-11-05:
So, we have following situation here:
1. If you put the latest version of bt-tether.py plugin wihich uses new
config on version 1.1.1 of pwnagotchi, you will get a "[WARNING] error
while loading
/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/bt-tether.py:
module 'pwnagotchi.plugins' has no attribute 'Plugin'" because this version
of plugin is incompatible with pwnagotchi version 1.1.1
2. If you will follow these instructions
https://pwnagotchi.ai/configuration/#bluetooth you will get an "[ERROR]
BT-TETHER: Please specify the mac in your config.yml" because bt-tether
plugin version that ships with version 1.1.1 is'nt compatible with new
style config.
As for now the solution would be either to use the old config style (
https://github.com/evilsocket/pwnagotchi/blob/v1.1.1/pwnagotchi/defaults.yml)
or try to manually update pwnagotchi from master.
Hope this will be fixed soon...

hackabean, 2019-11-05:
I found my solution!

Just a quick post to anyone facing similar problem in the future. In order to receive bt-tether connection make sure you power your pwnagotchi from the power micro usb slot and not the data one on the rpi0w!! Once you pair a device wait up to 5 minutes to get the connection back.

deveth0, 2019-11-06:
@yecukeng: you can upgrade to 1.2.1 which was released yesterday ;)

yecukeng, 2019-11-06:
@deveth0 yep, i can confirm it working with new style config on version 1.2.1

pincy, 2019-11-20:
On my 3rd attemp, the connection was initialized fine by the plugin.
Before the 3rd attemp, i made sure, to restart my android phone, activated tethering and removed the pwnagotchi before starting bluetooth.
Updating to the new version went flawless and the last message on my phones screen was, the neural network is ready.
But then i maybe got a little out of range? About 5-8meters away, saw the bt connection drop and immediatelly checked dmesg by the usb connection.

```
[ 1012.259480] ------------[ cut here ]------------
[ 1012.259528] WARNING: CPU: 0 PID: 1535 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 1012.259535] sysfs group 'byte_queue_limits' not found for kobject 'tx-0'
[ 1012.259539] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm raspberrypi_hwmon hwmon snd_timer snd i2c_bcm2835 spi_bcm2835aux bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem spi_bcm2835 bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq fixed uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 1012.259705] CPU: 0 PID: 1535 Comm: kbnepd bnep0 Tainted: G         C        4.19.66-Re4son+ #1
[ 1012.259710] Hardware name: BCM2835
[ 1012.259745] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 1012.259767] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 1012.259792] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 1012.259807] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 1012.259825] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 1012.259844] [<c021f370>] (sysfs_remove_group) from [<c0613460>] (netdev_queue_update_kobjects+0xc0/0x178)
[ 1012.259952] [<c0613460>] (netdev_queue_update_kobjects) from [<c0613578>] (netdev_unregister_kobject+0x60/0x80)
[ 1012.259977] [<c0613578>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 1012.259995] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 1012.260010] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 1012.260048] [<c05eb568>] (unregister_netdev) from [<bf5fb958>] (bnep_session+0x4d4/0x888 [bnep])
[ 1012.260100] [<bf5fb958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 1012.260116] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 1012.260123] Exception stack(0xdd8fdfb0 to 0xdd8fdff8)
[ 1012.260132] dfa0:                                     00000000 00000000 00000000 00000000
[ 1012.260143] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1012.260152] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1012.260159] ---[ end trace 21919dd0870a28c0 ]---
[ 1012.260265] ------------[ cut here ]------------
[ 1012.260292] WARNING: CPU: 0 PID: 1535 at /opt/kernel-builder_repos/Re4son/re4son-raspberrypi-linux/all/fs/sysfs/group.c:255 sysfs_remove_group+0x98/0xa4
[ 1012.260298] sysfs group 'power' not found for kobject 'bnep0'
[ 1012.260302] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse spidev brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm raspberrypi_hwmon hwmon snd_timer snd i2c_bcm2835 spi_bcm2835aux bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem spi_bcm2835 bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq fixed uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[ 1012.260457] CPU: 0 PID: 1535 Comm: kbnepd bnep0 Tainted: G        WC        4.19.66-Re4son+ #1
[ 1012.260462] Hardware name: BCM2835
[ 1012.260486] [<c0017e44>] (unwind_backtrace) from [<c0014d84>] (show_stack+0x20/0x24)
[ 1012.260506] [<c0014d84>] (show_stack) from [<c06f3278>] (dump_stack+0x20/0x28)
[ 1012.260529] [<c06f3278>] (dump_stack) from [<c0024624>] (__warn+0xf0/0x118)
[ 1012.260545] [<c0024624>] (__warn) from [<c002469c>] (warn_slowpath_fmt+0x50/0x70)
[ 1012.260561] [<c002469c>] (warn_slowpath_fmt) from [<c021f370>] (sysfs_remove_group+0x98/0xa4)
[ 1012.260588] [<c021f370>] (sysfs_remove_group) from [<c04a1474>] (dpm_sysfs_remove+0x58/0x60)
[ 1012.260607] [<c04a1474>] (dpm_sysfs_remove) from [<c0494e48>] (device_del+0x7c/0x360)
[ 1012.260622] [<c0494e48>] (device_del) from [<c0613594>] (netdev_unregister_kobject+0x7c/0x80)
[ 1012.260640] [<c0613594>] (netdev_unregister_kobject) from [<c05eb0f4>] (rollback_registered_many+0x1ec/0x508)
[ 1012.260655] [<c05eb0f4>] (rollback_registered_many) from [<c05eb4ac>] (unregister_netdevice_queue+0x9c/0x130)
[ 1012.260669] [<c05eb4ac>] (unregister_netdevice_queue) from [<c05eb568>] (unregister_netdev+0x28/0x30)
[ 1012.260703] [<c05eb568>] (unregister_netdev) from [<bf5fb958>] (bnep_session+0x4d4/0x888 [bnep])
[ 1012.260732] [<bf5fb958>] (bnep_session [bnep]) from [<c00432c0>] (kthread+0x120/0x160)
[ 1012.260746] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 1012.260753] Exception stack(0xdd8fdfb0 to 0xdd8fdff8)
[ 1012.260762] dfa0:                                     00000000 00000000 00000000 00000000
[... cut ...]
[ 1012.261283] [<c00432c0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[ 1012.261290] Exception stack(0xdd8fdfb0 to 0xdd8fdff8)
[ 1012.261298] dfa0:                                     00000000 00000000 00000000 00000000
[ 1012.261309] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1012.261318] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1012.261324] ---[ end trace 21919dd0870a28c2 ]---
```

Nothing helped, i had to reboot by ssh. But after the reboot it connected to my phone again.

sayak-brm, 2019-11-25:
@hackabean please close the issue if resolved. I've tested `bt-tether` on the v1.3.0 version with an Android phone and everything is working perfectly.

copysolo, 2019-12-01:
Pi zero waveshare_1:
iOS phone 13.2.3:
Pc hôte Windows 10:
hello, thank you for this project really nice.  I have a problem my unit and connect to my iphone but I have no internet signal.  My sharing and active.  can you tell me what is wrong.  I use the default config. Thx

H4ck3r404, 2019-12-08:
I have similar problem - my pwnagotchi connects to the phone and immedieatly disconnects. An addition it only works for the first pair. I wasn't able to have stable connection even so my mac adress was trusted and paired. Any ideas how could this be fixed?

copysolo, 2019-12-08:
> I have similar problem - my pwnagotchi connects to the phone and immedieatly disconnects. An addition it only works for the first pair. I wasn't able to have stable connection even so my mac adress was trusted and paired. Bluetoothctl output after typing: connect says that it failed. Any ideas how could this be fixed?
delete mac address with the remove 'mac address' command and restart the pairing, it worked for me

copysolo, 2019-12-08:
> I have similar problem - my pwnagotchi connects to the phone and immedieatly disconnects. An addition it only works for the first pair. I wasn't able to have stable connection even so my mac adress [...]

delete mac address with the remove 'mac address' command and restart the pairing, it worked for me

H4ck3r404, 2019-12-08:
Ok. I will try it tomorrow. Also I'm on Android 7 with MIUI rom. I don't know about it's bluetooth configuration. (Only thing that can be changed is switching bluetooth on and off.)

H4ck3r404, 2019-12-08:
For now thanks for advice. :)

copysolo, 2019-12-08:
> For now thanks for advice. :)
with pleasure, :D  last tip check well that the btnap and properly install.

H4ck3r404, 2019-12-10:
Nope. It is still broken. Where and how should I install btnap?

H4ck3r404, 2019-12-10:
After trying to confriming pair it outputs:

[CHG] Device <mac> Modalias: bluetooth:v00Fp1200d1436
[CHG] Device <mac> UUIDs: <uuid>
[CHG] Device <mac> ServiceResolved: yes
[CHG] Device <mac> Paired: yes
Pairing successful
[CHG] Device <mac> ServiceResolved: no
[CHG] Device <mac> Paired: no

H4ck3r404, 2019-12-10:
Pwnagotchi is not recognized as network sharing device. (Even that bt-tether is on.) Instead my phone recognizes it as numbers and contacts sharing device. Idk if it's a bug or something other.

copysolo, 2019-12-11:
> Nope. It is still broken. Where and how should I install btnap?

https://github.com/evilsocket/pwnagotchi/issues/377
watch here

operationairstrike, 2024-01-16:
Was there a solution? @hackabean