evilsocket/pwnagotchi #516: Advertise BLE Services for Better iOS/Android Support

nikhiljha, 2019-11-04:
**Is your feature request related to a problem? Please describe.**
Connecting the Pwnagotchi to an Android or iOS device is necessary for anything involving internet to work, but it's a huge pain. Even when connected, it's not possible to use the phone's GPS (for example) to determine position, which is a waste since you need to carry the phone around anyway.

**Describe the solution you'd like**
1. Update bluez to >= 5.4X if it isn't already.
2. Create a plugin that runs a BLE GATT server that allows (a) sending pcaps (b) requesting GPS coordinates (c) receiving GPS coordinates.

Once that's done, it's fairly trivial to create an app to interface with that. This is what a lot of devices (especially smartwatches) already do, so it should work.

**Additional context**
I'm imagining an app that allows you to connect to the pwnagotchi and send GPS coordinates to it when the pwnagotchi requests it. Then, when a pcap is generated, it gets sent back to the app for upload / processing / whatever. It should feel a lot nicer than BT tethering.

nikhiljha, 2019-11-04:
I can make a pretty app (Flutter / cross platform) once the GATT profiles are made, and I'm pretty sure the plugin bit isn't hard, but I want to gauge if this is something people would be interested in first.

evilsocket, 2019-11-04:
What is a huge pain exactly? Because it's pretty easy and it works smoothly out of the box ... i have the feeling you're creating issues where there aren't none.

evilsocket, 2019-11-04:
i think the top priority in terms of making an app is https://community.pwnagotchi.ai/t/pwnmail-web-client/46/5

nikhiljha, 2019-11-04:
That's a fair point. It's mostly an iOS thing (you need to have a WiFi AP on along with a Bluetooth AP). A GATT server is the more "correct" way of doing things since nearly every other device does it (Fitbit, etc.).

Keeping it as-is and building an app on top is definitely easier as things are right now though.

operationairstrike, 2024-01-16:
I need this improvement now!

operationairstrike, 2024-01-16:
@evilsocket pls implement

maxxer, 2024-01-16:
> @evilsocket pls implement

That's a very bad attitude. If you **need it now** then fund the development, don't expect others to do it for you, for free.

operationairstrike, 2024-01-16:
IM NOT A FUCKING DEV @maxxer

maxxer, 2024-01-16:
Then simply pay one. And also pay a therapist so you can learn how to behave.

operationairstrike, 2024-01-16:
Haha

evilsocket, 2024-01-16:
@maxxer thank you.
@operationairstrike che cazzo te ridi, a cojone?