evilsocket/pwnagotchi #521: UPS_LITE display info not clearing[BUG]

miotislucifugis, 2019-11-04:
**Describe the bug**
I just received and installed my ups.lite and am finding that the battery info at the top of the screen is not clearing... new characters are drawn on top of the old ones.  This is happening on both the waveshare screen and on the web ui.    A reboot clears things (obviously) but just until the data/numbers change.    The percentage part seems to be working properly... it appears to be the bold UPS text and the voltage that are affected.

Also... not sure if you are aware,  ups_lite is no longer included in the plugin section of default.yml.      Enabling it in config.yml  still seems to work tho, aside from the old characters not clearing on screen refreshes    (meaning it doenst crash when booting... )  

Using pwnagotchi v1.1.1. ,  using Waveshare v2  & usp-lite, 
i2c enabled in ifconfig.

dwi, 2019-11-05:
Workaround: I solved this by removing the voltage from the information and shifting the indicator to the right. You can do the following changes in the usp lite plugin 

`def on_ui_setup(ui):
    ui.add_element('ups', LabeledValue(color=BLACK, label='UPS', value='0%/0V', position=(ui.width() / 2 + 15, 0),
                                       label_font=fonts.Bold, text_font=fonts.Medium))`

`def on_ui_update(ui):
    ui.set('ups', "%2i%%" % (ups.capacity()))`

evilsocket, 2019-11-05:
can i have a screenshot so i can understand what we're talking about?

miotislucifugis, 2019-11-05:
Sorry, my original post is a little unclear. basically the BT plugin and the UPS plugin are writing to the same place on the screen.    DWi's workaround solved it.  here is an image:

[image]

spectrophagus, 2019-11-10:
Just another vote for this issue... can confirm the @dwi workaround fits nicely between BT status and uptime.

evilsocket, 2019-11-11:
I need a screeshot of the new layout

dwi, 2019-11-11:
This is how it looks with the change applied

[image]

commit dfd534ac (2019-11-11): fix: fixed ups_lite plugin layout (closes #521)
commit 52b40f04 (2019-11-12): Small UPS Lite typo fix

Fixing https://github.com/evilsocket/pwnagotchi/commit/dfd534ac41ebb907a8a1e3d921bfeb01919cd677 that fixes #521
closed 2019-11-11 by evilsocket