evilsocket/pwnagotchi #530: mesh protocol is not signed

evilsocket, 2019-11-05:
while developing the mesh protocol, i also implemented the signature for the payload so that impersonation/spoofing and in general sending fake data can't be done (relevant for things like #529) ... the idea is that each unit has a list of public keys of friendly units that are authorized to receive signed mesh data from. 

The current implementation is however commented:

https://github.com/evilsocket/pwngrid/blob/master/mesh/peer.go#L147
https://github.com/evilsocket/pwngrid/blob/master/mesh/peer.go#L199

because with a signature the payload would become too big for a single frame, and the injection would fail here:

https://github.com/evilsocket/pwngrid/blob/master/mesh/packet_muxer.go#L107

Sending more than one frame is not doable as the interface is hopping unpredictably on the wifi channels.

Ideally we should find a way to use a signature scheme that would generate a small overhead in size, maybe something like [BLS](https://medium.com/chain-intelligence/the-bls-signature-scheme-a-short-intro-801c723afffa).

ChipWolf, 2019-11-05:
ISO/IEC 9796 specifies a way to embed the message in the signature for recovery, if we want to reuse the existing RSA keypair.

bensmith83, 2019-11-09:
I had done something similar a long time ago (stuffing command and control data in the beacon's "Rates" field) - and had the sender send an 'ready to send' (i'm here, stop and listen to meeeeeee), then the listener can pause on that channel, send an 'ack' (i'm waiting around for a few), and then the sender can send the data in as many packets as necessary, maybe a '1 of n' bit set in the start of the data so the listener can validate. Seems like reinventing TCP on top of Beacons, but, also seems plausible.

evilsocket, 2019-11-09:
yes i tried that approach (with sync and sequence numbers), the problem is that in our case the wifi is hopping and we can't block that because of the advertisement