evilsocket/pwnagotchi #532: The 213d does not display the memtemp data in the correct place [BUG]

ian-rubenstein-mpf, 2019-11-05:
**Describe the bug**
The memtemp data is running over the edge of the screen

**To Reproduce**
Steps to reproduce the behavior:
1. load v1.2.1
2. enable memtemp (vertical or horizontal)
3. load UI

**Expected behavior**
Should be in the same location on screen as prior to update

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version 1.2.1
 - OS version osx, n/a
 - Type of hardware rpi0w, waveshare_213d
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

Arttumiro, 2019-11-05:
According to RustyPaper on slack same happens with Inky too, once switched to waveshare_2 everything looks perfect

srcr, 2019-11-08:
[image]

**Environment**
Pwnagotchi version 1.2.1
Type of hardware rpi0w, inkyphat

**note**
Just my 2 cents just upgraded to 1.2.1, and found the issue as described.

srcr, 2019-11-08:
<a href="https://github.com/evilsocket/pwnagotchi/issues/532" class="u-in-reply-to"></a>[image]

**Environment**
Pwnagotchi version 1.2.1
Type of hardware rpi0w, inkyphat

**note**
Just my 2 cents just upgraded to 1.2.1, and found the issue as described.

charlesrocket, 2019-11-08:
Same on `1.2.1` with inkyphat-red (rpi0w).

charlesrocket, 2019-11-09:
@ian-rubenstein-mpf could you drop `213d` from the title since its not just for waveshare case?

b33rANDfr13s, 2019-11-10:
memtemp was configured on my first 60+ hours or runtime in the last week.
[image]

I just configured auto-update today and updated to pwnagotchi v1.2.1

After updating, both horizontal and vertical memtemp displays are partially off-screen.

charlesrocket, 2019-11-10:
btw this might be an offtop but i think you should also blank your device names?) https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/grid.py#L12-L41 in case your device found internet over somewhere and you forgot to turn off `grid`

closed 2019-11-15 by evilsocket