evilsocket/pwnagotchi #539: Waveshare 2.9 inch missing from utils.py. Gets reported as unsupported display at boot on v1.2.1

demolition22, 2019-11-06:
The added Waveshare 2.9 inch in v1.2.0 did not get added to /pwnagotchi/utils.py

## Expected Behaviour

Display gets reported as Waveshare29inch

## Current Behaviour

Display gets reported as not supported 

## Possible Solution

Known solution: add 

`elif config['ui']['display']['type'] in ('ws_29inch', 'ws29inch', 'waveshare_29inch', 'waveshare29inch'):
        config['ui']['display']['type'] = 'waveshare29inch'`

## Steps to Reproduce (for bugs)

1.
2.
3.
4.

## Context

## Your Environment

- [x ] You're using the official images

- x[ ] You're using a raspberry pi 0

- [x ] You're using a supported LCD

3top1a, 2019-11-06:
could you help me fix this in a flashed env.?

3top1a, 2019-11-06:
also

you can reproduce this bug by adding "waweshare29inch" into the config, /usr/pwnagotchi launcher is going to fail to start since it doesn't recognize it.

demolition22, 2019-11-06:
> also
> 
> you can reproduce this bug by adding "waweshare29inch" into the config, /usr/pwnagotchi launcher is going to fail to start since it doesn't recognize it.

I'm not sure if i understand your question, when adding waveshare29inch to the config pwnagotchi still starts and works normally through the WebUI. However the connected 2.9 inch display does not function and in the logs it is reported as not a supported display since the /pwnagotchi/utils.py file does not have the option in the if statement for the 29inch. 

In order to fix it all I had to do was add the ELIF statement so that it would recognize and report that the waveshare29inch was set in the config.

demolition22, 2019-11-06:
Change has already been committed.

closed 2019-11-06 by demolition22