evilsocket/pwnagotchi #540: Whitelist not working [BUG]

elipsius, 2019-11-06:
**Describe the bug**
Whitelist no matter how entered is not used, my SSID is still pwned.  

**To Reproduce**
Steps to reproduce the behavior:
1. Launch with whitelist enabled and provide SSID list
2. wait
3. It will start pwning the whitelisted SSIDs and pcap files will appear.

**Expected behavior**
Should not pwn whitelist SSIDs
Whitelisted SSID pcap files should not show up

**Environment (please complete the following information):**
 - Pwnagotchi version 1.1.1
 - pwnagotchi image download 1.1.1
 - Raspberry Pi 3+
 - Google WiFi mesh is being whitelisted

Generic42, 2019-11-06:
I'm also seeing this behavior.

I've attempted the following white list formats
First attempt:

main:
  name: 'pwnagotchi'
  whitelist:
    - 'YourHomeNetworkMaybe'
    - 'Your  Second home Maybe'
  plugins:
    grid:
      enabled: true
      report: true
      exclude:
    - 'YourHomeNetworkMaybe'
    - 'Your  Second home Maybe'

Other attemp
main:
  name: 'pwnagotchi'
  whitelist:
    - YourHomeNetworkMaybe
    - Your  Second home Maybe
  plugins:
    grid:
      enabled: true
      report: true
      exclude:
    - YourHomeNetworkMaybe
    - Your  Second home Maybe

nucow69, 2019-11-07:
I noticed that too. Then I saw that the FAQ on pwnagotchi.ai says:
Your access points handshakes get still captured. This only prevents the deauth of the clients.

elipsius, 2019-11-07:
I'm doing this:

dadav, 2019-11-07:
Guys, like @nucow69 this is an expected behaviour. The whitelisted APs won't get "pwned" aka. deauthed, but captured. This is not a bug.

evilsocket, 2019-11-07:
As per docs:

**Your access points handshakes get still captured. This only prevents the deauth of the clients.**

closed 2019-11-07 by evilsocket

fennectech, 2019-12-27:
This should be mentioned on pwnagotchi.ai   Thank you for the info :3

RussellTaylor83, 2020-12-15:
Adding this to the docs would make a lot of sense. I'll see if I can add it and get it past PR.

funkymonkey5150, 2020-12-23:
It is mentioned on there