evilsocket/pwnagotchi #541: GPS not pulling data

elipsius, 2019-11-06:
I have gpsd configured with USB based GPS receiver, test using cgps and data pulls and shows location data and time correctly.  When I enable it in my config the json files show up but location. Data is has lats and longs always 0.

Not sure what I'm missing or of this is a bug.

elipsius, 2019-11-07:
My configuration:

```
#
main:
    name: 'manigotchi'                                                                                                                      plugins:
     gps:
            enabled: true
            speed: 19200
            device: /dev/ttyUSB0
```

evilsocket, 2019-11-07:
we need logs

elipsius, 2019-11-07:
Will pull them when I get a chance today

benleb, 2019-11-13:
pobably this was due to the broken gps plugin, fixed since https://github.com/evilsocket/pwnagotchi/commit/f4b886cf7bbf12e065307d04394a39b94ec74022 :) should work in next release