evilsocket/pwnagotchi #542: [BUG] plugin AircrackOnly - it seems to delete all the new handshakes

zenzen666, 2019-11-06:
**Describe the bug**
I've ran the plugin for one week with v1.1.1 on a pi0 and I had the 99% of all new handshakes deleted.
I haven't checked by hand if all the new handshakes was really unvalid but it seems unlikely to me.

Is it a good idea to rely on aicrack to check if a pcap is valid??

**Environment (please complete the following information):**
 - Pwnagotchi version 1.1.1
 - OS version
 - Type of hardware pi0

FrixosTh, 2019-11-06:
I believe this issue has already been resolved in v1.2.1 (v1.2.1 has a new issue which prevents AircrackOnly plugin to load - This should be resolved in the next release)

evilsocket, 2019-11-07:
ok i'm done with this plugin, i'm removing it

evilsocket, 2019-11-07:
it is just dumb to rely on aircrack only *as i always said*.

FrixosTh, 2019-11-07:
I really liked the concept of the plugin and I would love to see it stay.. as I mentioned on my previous comment, all issues up to now should be resolved in the next release.. but it's your project and the final decision...

evilsocket, 2019-11-07:
I'm just moving some of the plugins in a separate repositories of user contributed plugins, so we don't have to fix issues created by broken plugins in the core.

Also, let me reiterate: relying on aircrack only to filter handshakes **will delete a bunch of good pcap files that you might crack with hashcat**.

The assumption that aircrack can properly filter that stuff **is wrong**, it misses a lot of stuff.

closed 2019-11-07 by evilsocket

evilsocket, 2019-11-07:
see https://community.pwnagotchi.ai/t/announcement-some-of-the-plugins-have-been-moved-to-a-separate-repository/234

commit 2ea8e7fe (2019-11-07): misc: moved non-core and problematic plugins to separate repository (closes #542)