evilsocket/pwnagotchi #543: [BUG] Serial instead Ethernet interface

happostroph, 2019-11-06:
**Describe the bug**
On a fresh install of my pwnagotchi, I plugged it for the first time, but no new Ethernet interface appear.
With some investigations, I discover a serial USB (COM4 for me) that appear when I plug the rpi0.
Is it related to #159 ?

**To Reproduce**
Steps to reproduce the behavior:
1. Flash SD.
2. Create config.yml and edit it with the good properties.
3. Plug the SD crad to the RPI0.
4. Plug the RPI0 on Windows.

**Screenshot**

[image]
Here is the serial USB.

**Expected behavior**
I expect a new ethernet interface for my pownagotchi

**Environment:**
 - Pwnagotchi version : 1.2.1
 - OS version : Windows 10 1903
 - Type of hardware : RPI0WH + waveshare_2 + SD card SanDisk UHS-1
 - Any additional hardware used : mini usb - usb cable which allow data transfert

Thank you in advance !

PeachMangoes, 2019-11-07:
here try this one.
just update then that serial port using the driver below.

got it from this [site](https://www.factoryforward.com/pi-zero-w-headless-setup-windows10-rndis-driver-issue-resolved/)

[mod-duo-rndis.zip](https://github.com/evilsocket/pwnagotchi/files/3817781/mod-duo-rndis.zip)

mrseeker, 2019-11-07:
I had this same issue, fixed it by installing the RNDIS driver. Known bug, should be referenced here: https://pwnagotchi.ai/configuration/#connect-to-your-pwnagotchi

happostroph, 2019-11-07:
Well thanks guys ! This works perfectly.
I followed the steps from your [site ](https://www.factoryforward.com/pi-zero-w-headless-setup-windows10-rndis-driver-issue-resolved/) @PeachMangoes 

Yes @mrseeker , this could help many new users

closed 2019-11-07 by happostroph

hallo545401, 2020-11-13:
@PeachMangoes helped me as well, should really be mansioned on the config page