evilsocket/pwnagotchi #549: [BUG] Failure to initialize display (WSv2) after 1.2.1 update

audibleblink, 2019-11-07:
**Describe the bug**
Pwnagotchi fails to start

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.2.1 to sdCard
2. Attach WaveshareV2 e-paper display
3. Connect Power

**Expected behavior**
Normal boot and e-paper display updates

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.2.1
 - OS version: Linux pwnagotchi 4.19.66-Re4son+ #1 Sun Aug 18 13:42:02 AEST 2019 armv6l GNU/Linux
 - Type of hardware: Pi Zero v 1.1
 - Any additional hardware used: Waveshare v2

**Additional context**

This is the debug output when attempting to start `pwnagotchi`. Stuck here after 1 hour.

```
root@pwnagotchi:/home/pi# systemctl stop pwnagotchi
root@pwnagotchi:/home/pi# pwnagotchi --debug --manual
[2019-11-07 15:24:22,441] [DEBUG] loading plugins from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default - enabled: ['grid', 'auto-update']
[2019-11-07 15:24:22,451] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2019-11-07 15:24:22,457] [DEBUG] loaded plugin grid as <grid.Grid object at 0xb4df3430>
[2019-11-07 15:24:22,461] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2019-11-07 15:24:25,108] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0xb4df3530>
[2019-11-07 15:24:25,112] [INFO] grid plugin loaded.
[2019-11-07 15:24:25,115] [INFO] [update] plugin loaded.
[2019-11-07 15:24:25,128] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-11-07 15:24:25,132] [INFO] initializing waveshare v2 display

```

Upon pressing Ctrl+C, the following stack trace occurs:

```
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 50, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__      
    self.init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 69, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/waveshare2.py", line 60, in initialize
    self._display.init(self._display.FULL_UPDATE)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 175, in init
    self.wait_until_idle()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 161, in wait_until_idle
    delay_ms(100)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 72, in delay_ms
    time.sleep(delaytime / 1000.0)
```

Removing the screen and restarting allows for normal operation.

gagna, 2019-11-07:
Please post your config.yml

audibleblink, 2019-11-07:
Stock (empty) config.yml, so it's using default.yml's settings

audibleblink, 2019-11-07:
Appears to be a bad solder job on the waveshare on the HAT connector. I wired it manually with 8-pin connector that came with it and everything worked normally. Closing as not software related.

closed 2019-11-07 by audibleblink