evilsocket/pwnagotchi #550: Enable TRIM for SDcard lifetime increase

Spindel, 2019-11-07:
**Is your feature request related to a problem? Please describe.**
I'm always annoyed when I need to regularly reflash cheapish SD-cards because debian ships without TRIM timers by default, causing a failure in internal wear levelling and making them corrupt the filesystems without any data issues.

**Describe the solution you'd like**
enable the fstrim.timer by default

**Describe alternatives you've considered**
Discard mount option, but it adds a noticable write delay for everything, which isn't fun.

**Additional context**
Small feature fix for all SD-cards that support it.

xenDE, 2019-11-07:
"noatime" for rootfs is a good idea too, imho

xenDE, 2019-11-07:
> "noatime" for rootfs is a good idea too, imho

sorry, already set

closed 2019-11-08 by evilsocket