evilsocket/pwnagotchi #555: [BUG] screen_refresh with waveshare_2 break display

darknao, 2019-11-08:
**Describe the bug**
The screen start showing garbage, or even completely crash, when using the screen_refresh plugin on the waveshare v2

**To Reproduce**
Steps to reproduce the behavior:
1. enable the screen_refresh plugin
2. wait for a full refresh

**Expected behavior**
The screen should do a full refresh, and continue to work afterward.

**Environment (please complete the following information):**
 - 1.2.1
 - pi0w
 - waveshare v2

**Additional context**
Waveshare v2 use partial refresh all the time.
According to Waveshare specs, you should do a full refresh once in a while when using partial mode to avoid ghosting and burn in issue.
Additionally, in my case, I sometimes get quite severe ghosting + black dots in the background, and using the screen_refresh plugin seems to help. 
PR about this will follow.

evilsocket, 2019-11-08:
that is why the plugin has been moved in [a separate repo](https://github.com/evilsocket/pwnagotchi-plugins-contrib) that says **use them at your own risk**.

https://community.pwnagotchi.ai/t/announcement-some-of-the-plugins-have-been-moved-to-a-separate-repository/234

evilsocket, 2019-11-08:
(also there's no real need of full screen refreshes on v2)

closed 2019-11-08 by evilsocket

darknao, 2019-11-08:
not according to specs : https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT 

> Note that you cannot use Partial refresh all the time, you should full refresh e-paper regularly, otherwise, ghost problem will get worse even damage.

evilsocket, 2019-11-08:
not according to months of testing and our customized codebase, which is different than their examples and sdk

JrdBrd, 2019-12-13:
i get the same severe ghosting and black dotted background after prolonged uptime, maybe it is worth a look into