evilsocket/pwnagotchi #557: Add link to Pwngrid page to web interface

jabella, 2019-11-08:
I've got a few units, but I don't always know the pwnfile / address for each when I'm away from a computer.  For example - one of my units is:

11bbc31713856644b929bb8361a02ba470899febac79fb193ba18d42800a748a

- But I don't know my other ones without connecting to a laptop and checking.  

**Describe the solution you'd like**

Can we get a link on the web interface that links right to the pwngrid page for that unit?  Like "Open my Pwngrid page" that links to the relevant page for that unit:

https://pwnagotchi.ai/pwnfile/#11bbc31713856644b929bb8361a02ba470899febac79fb193ba18d42800a748a

budd3993, 2019-11-08:
A page showing peers with links to their pwnfile pages would be useful too.  Maybe pull some stats/info from pwngrid api too

evilsocket, 2019-11-08:
ooo i like where this is going :D will start working on this asap ... maybe it can be the start of https://community.pwnagotchi.ai/t/pwnmail-web-client/46

commit a2ac6794 (2019-11-10): new: pwnfile link in the web ui (closes #557)
closed 2019-11-10 by evilsocket