evilsocket/pwnagotchi #560: Adding plugin event on shutdown

darknao, 2019-11-08:
**Describe the solution you'd like**
Something like **on_shutdown** event allowing plugins to do one last action before going to sleep. 
Like backup, handshakes upload, generating last session statistics, clearing the screen...

Arttumiro, 2019-11-08:
on_shutdown already does stuff, like makes the screen go to sleep mode (sleeping face and a good night message)

darknao, 2019-11-08:
True. But it's not a plugin event. 
The on_shutdown name was maybe misleading. It's more a "shutdown" event that can be used by plugins with the "on_shutdown" method, if that make sense.

xenDE, 2019-11-11:
on_shutdown() needs to throw a on_before_shutdown() event.

so plugins can do things before.

cdiemel, 2019-11-28:
I just came searching for this specific thing.  What is the status of this?