evilsocket/pwnagotchi #583: [BUG] 5G channels above 140 aren't supported

andrewbeard, 2019-11-12:
**Describe the bug**
Despite hardware support pwnagotchi never uses wifi channels above 140

**To Reproduce**
Steps to reproduce the behavior:
1. Install pwnagotchi 1.2.1 on a Raspberry Pi 4 (or Pi 3 with an external adapter that supports 5g)
2. Run iw list:
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (20.0 dBm)
			* 5190 MHz [38] (20.0 dBm)
			* 5200 MHz [40] (20.0 dBm)
			* 5210 MHz [42] (20.0 dBm)
			* 5220 MHz [44] (20.0 dBm)
			* 5230 MHz [46] (20.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (radar detection)
			* 5280 MHz [56] (20.0 dBm) (radar detection)
			* 5300 MHz [60] (20.0 dBm) (radar detection)
			* 5320 MHz [64] (20.0 dBm) (radar detection)
			* 5500 MHz [100] (20.0 dBm) (radar detection)
			* 5520 MHz [104] (20.0 dBm) (radar detection)
			* 5540 MHz [108] (20.0 dBm) (radar detection)
			* 5560 MHz [112] (20.0 dBm) (radar detection)
			* 5580 MHz [116] (20.0 dBm) (radar detection)
			* 5600 MHz [120] (20.0 dBm) (radar detection)
			* 5620 MHz [124] (20.0 dBm) (radar detection)
			* 5640 MHz [128] (20.0 dBm) (radar detection)
			* 5660 MHz [132] (20.0 dBm) (radar detection)
			* 5680 MHz [136] (20.0 dBm) (radar detection)
			* 5700 MHz [140] (20.0 dBm) (radar detection)
			* 5720 MHz [144] (20.0 dBm) (radar detection)
			* 5745 MHz [149] (20.0 dBm)
			* 5765 MHz [153] (20.0 dBm)
			* 5785 MHz [157] (20.0 dBm)
			* 5805 MHz [161] (20.0 dBm)
			* 5825 MHz [165] (20.0 dBm)
3. Check the supported channels in /var/log/pwnagotchi.log:
[2019-11-11 16:36:33,426] [INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140]

**Expected behavior**
The list of supported channels via pwnagotchi should match those supported by the hardware. Eventually my access point on channel 161 should be observed.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.2.1
 - Linux pwnpi4 4.19.81-Re4son-v7l+ #1 SMP Wed Nov 6 13:10:35 AEDT 2019 armv7l GNU/Linux
 - Raspberry Pi 4 with built in wifi or Pi3 with RT3572 adapter

**Additional context**
I've tried change the reg country by running `iw reg set US` to no effect

mrseeker, 2019-11-12:
Supported channels are found using "/sbin/iwlist wlan0 freq"

From the man page:
```
freq[uency]/channel
Give the list of available frequencies in the device and the number of defined channels. Please note that usually the driver returns the total number of channels and only the frequencies available in the present locale, so there is no one-to-one mapping between frequencies displayed and channel numbers.
```

evilsocket, 2019-11-12:
@mrseeker no @andrewbeard here [is right](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ai/gym.py#L40) ... check the [hard limit here](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/ai/featurizer.py#L7) and [here](https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/mesh/wifi.py#L1).

evilsocket, 2019-11-12:
So the main problem now is that if i change that 140 directly, i basically break backwards compatibility with all the pretrained brain.nn files ... I'll have to have to check which supported channels are available and only use the wider spectrum for 5g

evilsocket, 2019-11-12:
oh i just realized this is easier than i thought thanks to that very python expression, nothing will change for 2.4 adapters :D

closed 2019-11-13 by evilsocket

Esp8266Alfa, 2019-11-19:
I have a silly question, just buy a dual band adapter to plug in and use?

gerard780, 2019-11-20:
Is this bug truly fixed? I'm still struggling to see channels > 140 on my pi4. It seems to me that the command used in utils.c (iwlist wlan0 freq) only returns the first 32 channels regardless of what the country code is set to. Using "iw phy0 channels" returns 39 available channels. The 32 channel limit seems to be part of iwlist as it happens on every system I've tested. 

Get current country: # iw reg get
Compare the output of these two lines. 
$ iwlist wlan0 freq
$ iw phy0 channels | grep " MHz" | grep -v disabled

commit a8c07ba9 (2019-11-13): fix: supporting channels greater than 140 for 5g (closes #583)