evilsocket/pwnagotchi #585: Unable to access using Putty in Windows

RChadwick7, 2019-11-12:
I have my computer set as 10.0.0.1, and I can see the web interface at 10.0.0.2, but I can't SSH to 10.0.0.2 using Putty. I get a 'Software caused connection abort' message.

mrseeker, 2019-11-12:
First a couple of questions:
1. Is the device still booting? As in, is the LED flashing?
2. What does "ipconfig /all" return? Your RNDIS interface should return 10.0.0.1 as IP.
3. What does "route print" return? Anything starting with 10.0.0.0 should point to the RNDIS interface.

closed 2019-11-12 by evilsocket

RChadwick7, 2019-11-12:
Thanks for the response!
It's finished booting, and working. Web address works, but only with 10.0.0.2, not pwnagotchi.local

ipconfig /all returns 10.0.0.1 as IP, 255.255.255.255 subnet, 0.0.0.0 gateway

Route Print returns:
10.0.0.0   255.255.255.0   On-Link   10.0.0.1
10.0.0.1   255.255.255.255   On-Link   10.0.0.1