evilsocket/pwnagotchi #588: Pwnagotchi login

aliforyou, 2019-11-12:
I use raspberry p zero w with my pc screen by hadmi to vaga adapter,  I use the last version of pwnagotchi image and I followed the instructions to buid a  pwnagotchi everything was  fine  but I don't know what is the pwnagotchi login I tried raspberry default user but nothing happen it keep tell me that login is incorrect 
[image]
Sorry for my bad English

Arttumiro, 2019-11-12:
pi/raspberry
login is pi, password is raspberry
Use passwd after boot to change password to something else

closed 2019-11-12 by evilsocket

spryder89, 2024-02-17:
hi 
i have the same issue. 
But the login isnt pi/raspberry on my device, can you help me?