evilsocket/pwnagotchi #591: [BUG] Unit is stuck while initializing display

phantasmthewhite, 2019-11-12:
**Describe the bug**
RPI is booting up, display doesn't refresh, while going through the logs it says "Initializing waveshare v2 display" and that's it.
As of yesterday everything was working fine, but at some point it stopped refreshing, now I can still access all functionnalities through SSH but no more screen..

**To Reproduce**
Steps to reproduce the behavior:
1. Boot the pwnagotchi.
2. That's it.

**Expected behavior**
Booting up and refreshing the screen.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.2.1
 - OS version: Windows 10 / ParrotSec 4.7
 - Type of hardware: RPI Zero WH + Waveshare eInk v2

evilsocket, 2019-11-12:
it sounds like the screen is dead more than a bug ... did it break? did the unit hit somewhere?

phantasmthewhite, 2019-11-12:
Screen's integrity looks as good as new, no crash or anything, been in my confy jacket pocket all the time :/

mrseeker, 2019-11-12:
You can try to run the test app from waveshare, and see if that works?

https://github.com/waveshare/e-Paper

That way you can be sure that it is a broken screen and not the software itself.

lumir-mrkva, 2019-11-27:
@phantasmthewhite make sure that you have correct screen configured, and if you have 3 color waveshare v1 display, set color to something other than black

moheshmohan, 2019-12-17:
This is a problem is with waveshare displays. The busy pin on the display will keep on true indefinitely, the only workaround i was able to find was forcefully resetting it by turning off and on few times. I did fix this issue with tricolor display on this pull request #720 watch for the last of the commits on this pull request to see the freeze recovery routine. Currently this is for waveshare 2.13 inch tricolor variants (b & C) but same approach will work for others too, just see original waveshare sources for display power on and power off commands and change accordingly.

pipesmith, 2020-08-03:
Having this issue as well.  I have installed Waveshare display on a RPi4 and it operates perfectly.  My PiZero will boot fine and work fine without the screen.  Once the screen is installed, it won't boot past the "Initializing waveshare v2 display".  

Bettercap and SSH work, just not the WebUI and I believe the pwnagotchi service.  I've tried different cards, several config files, changing waveshare_2 to waveshare_1 and color red.  

[2019-07-10 04:18:23,080] [DEBUG] loading plugins from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default - enabled: ['grid', 'webcfg', 'auto-update', 'logtail', 'session-stats', 'led']
[2019-07-10 04:18:23,090] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2019-07-10 04:18:23,097] [DEBUG] loaded plugin grid as <grid.Grid object at 0xb53bc250>
[2019-07-10 04:18:23,101] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/webcfg.py
[2019-07-10 04:18:23,107] [DEBUG] loaded plugin webcfg as <webcfg.WebConfig object at 0xb53bc4d0>
[2019-07-10 04:18:23,111] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2019-07-10 04:18:23,121] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0xb53bc910>
[2019-07-10 04:18:23,125] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2019-07-10 04:18:23,132] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0xb53bcbf0>
[2019-07-10 04:18:23,136] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/logtail.py
[2019-07-10 04:18:23,142] [DEBUG] loaded plugin logtail as <logtail.Logtail object at 0xb53bceb0>
[2019-07-10 04:18:23,147] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/led.py
[2019-07-10 04:18:23,154] [DEBUG] loaded plugin led as <led.Led object at 0xb53ce650>
[2019-07-10 04:18:23,159] [DEBUG] loading plugins from  - enabled: ['grid', 'webcfg', 'auto-update', 'logtail', 'session-stats', 'led']
[2019-07-10 04:18:23,178] [INFO] [update] plugin loaded.
[2019-07-10 04:18:23,183] [INFO] Logtail plugin loaded.
[2019-07-10 04:18:23,187] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2019-07-10 04:18:23,185] [INFO] webcfg: Plugin loaded.
[2019-07-10 04:18:23,190] [INFO] grid plugin loaded.
[2019-07-10 04:18:23,202] [DEBUG] [led] event 'loaded' set
[2019-07-10 04:18:23,216] [INFO] Session-stats plugin loaded.
[2019-07-10 04:18:23,231] [DEBUG] [led] using pattern 'oo  oo  oo oo  oo  oo  oo' ...
[2019-07-10 04:18:23,354] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 04:18:23,357] [INFO] initializing waveshare v2 display

Haskar, 2023-11-04:
I know it's been aaaalmost 4 years since the topic was opened but I ran into the same issue today and thankfully I had two pwnagotchis with one working fine and the other one stuck in the "restart loop" where the waveshare display reinitializes over and over again. the logs where the same as in the OPs sreenshot. The only difference was that the not working pwnagotchi was running ~24/7 on a 16GB chard and the other one was barely turned on on a 32GB card. 

It turns out that the filesystem /var/tmp/pwnagotchi was 100% full. So I changed the config.toml entry
fs.memory.mounts.data.size = "10M"
to 
fs.memory.mounts.data.size = "1000M"
and rebooted the pwnagotchi. (restarting the service does not work, obviously)

hope that helps anyone in the future :)