evilsocket/pwnagotchi #592: Epochs run long when internet is not available

neurovish, 2019-11-12:
[pwnagotchi-test.log](https://github.com/evilsocket/pwnagotchi/files/3836535/pwnagotchi-test.log)
When there is no connectivity to the internet (specifically the API, https://api.pwnagotchi.ai), pwnagotchi runs very slowly (long epochs) with CPU usage under 50%.

Steps to reproduce the behavior:
1. Power on pwnagotchi with no internet
2. Wait for AI to load
3. Wait an hour or so
4. After about an hour of runtime, the logs should show long-running epochs ( > 20min) once the AI has loaded.

Expectation is the realtime length for an epoch to be far less than 20 minutes. With internet connectivity, epochs usually last less than 5 minutes.

Environment
- pwnagotchi v1.2.1
- pwngrid v1.10.1
- pi0w

Attached file shows output of "egrep -a "\[epoch|brain.nn" /var/log/pwnagotchi.log |tail -n 80". The first 33 epochs in this file cover almost 8 hours of runtime off of a battery with no internet connectivity. The next 6 epochs cover 43 minutes of runtime off of a USB connection with connection sharing blocked by the host.

Also of note is the pwngrid-peer.log, which mostly consists of lines that look like: 
2019-11-01 15:58:59 err POST https://api.pwnagotchi.ai/api/v1/unit/enroll (20.012588s) Post https://api.pwnagotchi.ai/api/v1/unit/enroll: dial tcp: lookup api.pwnagotchi.ai on 9.9.9.9:53: read udp 10.0.0.2:53978->9.9.9.9:53: i/o timeout

Arttumiro, 2019-11-12:
Wait this is the same thing i had happen to me a long time ago when i said i had like 36% cpu usage when i wait 45 min or more and then connect bt tether, yeah this is interesting

neurovish, 2019-11-12:
Yeah, same thing most likely. I tested on my laptop by not routing the pi to the internet, ran for a couple hours and went through about 4 epochs, then setup iptables to route the pi and instantly saw 10 epochs in about 15 minutes.

I haven't dug into the code/scripts yet to see what it is doing, mostly because I am not familiar at all with go.

evilsocket, 2019-11-12:
i think the problem is here https://github.com/evilsocket/pwnagotchi/blob/master/bin/pwnagotchi#L123 ... the grid plugin attempts to talk with the API blocking the loop ... fixing

evilsocket, 2019-11-12:
if you have a way to test from master, let me know if the fix works 👍

closed 2019-11-12 by evilsocket

neurovish, 2019-11-13:
Yes! I think it works. Been running for about 30 - 40 minutes with the internet link cut-off, and already on the 18th epoch and 2nd training epoch. Thanks man!

evilsocket, 2019-11-13:
great :D

commit 8fcfd4ca (2019-11-12): fix: plugin events dispatch is now asynchronous (fixes #592)