evilsocket/pwnagotchi #604: Include InkyFAST for faster updates to the screen for Inky pHat users

JRWR, 2019-11-13:
Requires testing, but someone made a hack to do faster partial updates to the display shown here:

https://github.com/wavelet2/inky_fast

Arttumiro, 2019-11-13:
Not completely sure but doesnt fastAndFurious already do this? but its not mentioned in default.yml comments

dipsylala, 2019-11-15:
Yeah, I know the source for this and I did some research around it when I was implementing fastAndFurious. There are some issues around this that I tried to consider when improving the Phat speed:
* Keep the display quality
* Prevent burn-in
* Keep the battery life

We've stripped down the LUT in a similar way to Wavelet did - I'm reluctant to keep the display powered up though. 

We're ignoring Reds to improve speed

I tried upping the SPI but got screen issues (garbage) on multiple units - this could be worth looking into a bit more.