evilsocket/pwnagotchi #610: [BUG] Changing interface in config.yml is not effective

andrewbeard, 2019-11-14:
**Describe the bug**
The main.iface option in config.yml appears to be for changing the interface the device uses for sniffing, but the mon0 and wlan0 are hardcoded in multiple scripts.

**To Reproduce**
Steps to reproduce the behavior:
1. Add main.iface to /etc/pwnagotchi/config.yml with a value of mon0
2. Bettercap still tries to use mon0
3. Pwngrid still tries to use mon0

**Expected behavior**
There should be a method to specify an alternate wireless interface when using other than the built-in wifi adapter.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.2.1
 - Linux pwnpi3 4.19.81-Re4son-v7+ #1 SMP Wed Nov 6 10:16:47 AEDT 2019 armv7l GNU/Linux
 - Raspberry Pi 3 w/ an Alfa 5Ghz wireless adapter

**Additional context**
To use another wireless adapter the following scripts must be modified:
/etc/pwnagotchi/config.yml
/usr/bin/bettercap-launcher
/etc/systemd/system/pwngrid-peer.service
/usr/bin/pwnlib

In the event of a system update (like the 1.10.3 pwngrid update pushed recently) everything but the config.yml file can be overwritten with warning.

SchizoDuckie, 2019-11-14:
I had the same problem trying to configure some settings and plugins.
The trick is to save your modified config.yml to the BOOT partition, it will get installed on next boot.

This wasn't really intuitive to me without reading the docs thoroughly.

evilsocket, 2019-11-15:
and how is this a bug? it's not even related to this software given that, *for this software*, the main.iface config does work.

closed 2019-11-15 by evilsocket

andrewbeard, 2019-11-15:
Is there a seperate repo for the overall pwnagotchi device? I was under the impression that since the builder scripts were in here (and the config.yml file had config entries for bettercap) it covered not just the pwnagotchi process but the final config of the device.

As it stands right now there seems to be no way to reconfigure the pwnagotchi device as a whole to use a different wifi interface. Should I reopen this as an enhancement and if so which repo should it get tagged against?

evilsocket, 2019-11-15:
no there's no repo for the device as a whole, each component has a way to change the interface, so the problem you remarked just doesn't exist form my perspective ...

andrewbeard, 2019-11-15:
@evilsocket It sounds like that may be where the disconnect is coming in. Although each component has a way to change the interface, as a whole it doesn't work as a system. 

For example, I can change the interface used for bettercap on the command line as a standalone tool. That works for bettercap standalone but when used as part of pwnagotchi the interface is a part of /usr/bin/pwnlib. If you update pwnagotchi (either a backup, reflash of the card, and restore or updating via pip out of the git repo) pwnlib is overwritten back to the original wlan0/mon0 interface. pwnlib is part of the pwnagotchi repo so even though the pwnagotchi process isn't the problem (and is influenced by main.iface) when you set up pwnagotchi is has a tendency to overwrite stuff that gets used by everything else.

I guess from my perspective it seems weird that you'd want to change the interface used by the pwnagotchi process (via main.iface) without changing the interfaces for the other two processes at the same time. Sorry if this kind of seems rambling, I think I just have a different view of the pwnagotchi as a device rather than separate pieces of software as I don't use pwnagotchi/pwngrid/bettercap independently of each other.

bruvv, 2019-12-28:
I too tried everything without disable the raspberry pi's internal wifi card to get this to work. This should be some standard setting in config.yml.

I made a gist (https://gist.github.com/bruvv/f13cc2c74e52368bd32e1e8ef4e76b4d) that enables mon1 for wlan1 but pwnagotchi keeps enabling mon0 and keeps using that. There is no way in the world that I know of (even after reading the docs here: https://pwnagotchi.ai/installation/#installing-on-any-gnu-linux and here: https://community.pwnagotchi.ai/t/enable-an-external-wifi-card/163/15)
So no sure why you closed this but this is a bug indeed and perhaps someone can fix it...

evilsocket, 2019-12-28:
@andrewbeard your different view has been noted.

@bruvv it is not a bug because config.yml alone is not meant to apply the config system wide, but just for the pwnagotchi service, while there are other 2 services referencing the interface (bettercap and pwngrid). Bettercap is the one starting mon0, i didn't test your gist but if the changes are applied properly and everywhere is needed it just works, you can keep wlan0 and just use wlan1/mon1.

bruvv, 2019-12-28:
@evilsocket thanks for the quick reply. Perhaps can you check my gist? It is putting mon1 up. When I boot the pi, mon0 is created aswell. When I remove mon0 with: ```iw dev mon0 del``` it just reappears when I restart pwnagotchi. I cannot tell for sure if the pwnagotchi is using the mon1 interface but when I check with iwconfig the data that the interface is capturing I can see that mon1 is static and mon0 is receiving and sending data. So I assume mon0 is being used.